gnuradio/gnuradio #6102: Unknown CMake command "GR_SWIG_MAKE"

dxxx9, 2022-08-19:
### What happened?

Created a custom OOT module and followed the instructions starting with

```
cmake ../
```

This results in the error referring to SWIG (see log below).

(Developing the OOT in c++)

### System Information

OS: (Linux Distro, Mac, Windows)
GR Installation Method: (native distro packages - apt, rpm, etc., Conda, Source

### GNU Radio Version

3.10 (maint-3.10)

### Steps to Reproduce the Problem

Issue is the same for:

>OS: macOS 12.3
GR version: 3.10.3.0

and
>OS: Ubuntu 20.04
GR version: 3.9.5.0

### Relevant log output

```shell
CMake Error at swig/CMakeLists.txt:37 (include):
  include could not find requested file:

    GrSwig

CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".
```

mormj, 2022-08-19:
The swig directory in your oot could mean your oot isn't fully updated to 3.9/3.10
https://wiki.gnuradio.org/index.php?title=GNU_Radio_3.9_OOT_Module_Porting_Guide

That swig dir shouldn't be there if you are using GR 3.9/3.10

Was the OOT generated with `gr_modtool` when 3.8 is installed?  Is 3.8 fully uninstalled?

One other thing I vaguely remember being an issue is the `newmod_path` setting in the `[modtool]` section of the settings - at some point between installs this was set to an old installation

dxxx9, 2022-08-20:
@mormj Ah, thank you very much! I was not aware of this change.
As `gr_modtool` does not work on macOS (that's for another issue) I used a VM for running `gr_modtool` which had a version `< 3.9`. And yes, I noticed the wrong `newmod_path` entry too.

Below is only with `Ubuntu 20.04` and `gr v3.9.5`:
So instead of following the porting guide, I created a brand-new OOT module with v3.9. But when doing the build process, `cmake ../` fails with the following message

```shell
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrPybind.cmake:201 (message):
  Python bindings for <cpp-modul-block-header>.h are out of sync
Call Stack (most recent call first):
  python/bindings/CMakeLists.txt:34 (GR_PYBIND_MAKE_OOT)
```

Have you any clue what's going wrong here?

Has that something to do with the `pybind11` package?
I noticed [the hint](https://wiki.gnuradio.org/index.php?title=GNU_Radio_3.9_OOT_Module_Porting_Guide#Pybind11_Python_Bindings) in the porting guide that v2.5.0 is required and followed the described steps to build and install it manually.

willcode, 2022-08-22:
Please bring this up on chat.gnuradio.org or the mailing list. It looks like you need to update an MD5 hash for a PyBind11 file in your OOT module.

closed 2022-08-22 by willcode

dxxx9, 2022-08-22:
*---- Addendum, in case anyone has a similar problem. ----*

After a lot of debugging, I found that executing
```console
gr_modtool bind <block-name>
```
fixed the issue. But sadly, I'm not able to tell why it works now.

Neither the named files in the logs
```console
INFO Parsing source file "/media/psf/_vmshare/<module>/gr-<module>/include/<module/<cpp-block>.h"
Writing binding code to ./python/bindings/<cpp-block>_python.cc
Writing binding code to ./python/bindings/docstrings/<cpp-block>_pydoc_template.h
```
nor any other file was changed (compared untracked changes).

It would be interesting to know that to better understand how the process works.
If someone can tell me, I would be happy!