gnuradio/gnuradio #6103: gr-analog: fm_tx example generates invalid cpp code

dl1ksv, 2022-08-20:
### What happened?

The fm_tx example in gr-analog/examples should be able to generate cpp. But the cpp code generation fails with

[ 75%] Building CXX object CMakeFiles/fm_tx.dir/fm_tx.cpp.o
/home/schroer/gnuradiocomponents/bugs/fm_tx/fm_tx.cpp: In Konstruktor »fm_tx::fm_tx(int)«:
/home/schroer/gnuradiocomponents/bugs/fm_tx/fm_tx.cpp:263:36: Fehler: »irdes« wurde in diesem Gültigkeitsbereich nicht definiert
  263 |         std::vector<float> taps = {irdes.low_pass_2(nfilts, nfilts*in_rate, 20e3, 2e3, 60, window.WIN_KAISER};
      |                                    ^~~~~
/home/schroer/gnuradiocomponents/bugs/fm_tx/fm_tx.cpp:263:92: Fehler: ungültige Verwendung von Elementfunktion »QWidget* QWidget::window() const« (»()« vergessen?)
  263 |         std::vector<float> taps = {irdes.low_pass_2(nfilts, nfilts*in_rate, 20e3, 2e3, 60, window.WIN_KAISER};
      |                                                                                            ^~~~~~
      |                                                                                                  ()
/home/schroer/gnuradiocomponents/bugs/fm_tx/fm_tx.cpp:263:109: Fehler: »{<expression error>}« konnte nicht von »<brace-enclosed initializer list>« nach »std::vector<float>« umgewandelt werden
  263 |         std::vector<float> taps = {irdes.low_pass_2(nfilts, nfilts*in_rate, 20e3, 2e3, 60, window.WIN_KAISER};
      |                                                                                                             ^
      |                                                                                                             |
      |                                                                                                             <brace-enclosed initializer list>

**By the way**
The gr::analog::frequency_modulator_fc in the analog_wfm_tx.block.yml is no adequate replacement for the python block
analog.wfm_tx

### System Information

OS: Fedora 35
GR Build from latest Source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-211-g26082606

### Steps to Reproduce the Problem

Start grc
open gr-analog/examples/ fm_tx
Set output language to c++
Try to run the flowgraph

### Relevant log output

```shell
[ 75%] Building CXX object CMakeFiles/fm_tx.dir/fm_tx.cpp.o
/home/schroer/gnuradiocomponents/bugs/fm_tx/fm_tx.cpp: In Konstruktor »fm_tx::fm_tx(int)«:
/home/schroer/gnuradiocomponents/bugs/fm_tx/fm_tx.cpp:263:36: Fehler: »irdes« wurde in diesem Gültigkeitsbereich nicht definiert
  263 |         std::vector<float> taps = {irdes.low_pass_2(nfilts, nfilts*in_rate, 20e3, 2e3, 60, window.WIN_KAISER};
      |                                    ^~~~~
/home/schroer/gnuradiocomponents/bugs/fm_tx/fm_tx.cpp:263:92: Fehler: ungültige Verwendung von Elementfunktion »QWidget* QWidget::window() const« (»()« vergessen?)
  263 |         std::vector<float> taps = {irdes.low_pass_2(nfilts, nfilts*in_rate, 20e3, 2e3, 60, window.WIN_KAISER};
      |                                                                                            ^~~~~~
      |                                                                                                  ()
/home/schroer/gnuradiocomponents/bugs/fm_tx/fm_tx.cpp:263:109: Fehler: »{<expression error>}« konnte nicht von »<brace-enclosed initializer list>« nach »std::vector<float>« umgewandelt werden
  263 |         std::vector<float> taps = {irdes.low_pass_2(nfilts, nfilts*in_rate, 20e3, 2e3, 60, window.WIN_KAISER};
      |                                                                                                             ^
      |                                                                                                             |
      |                                                                                                             <brace-enclosed initializer list>
```

moatom, 2022-10-15:
I also encountered this type of problem when generating just a low-pass filtering code.
Aside from the obvious part, you can fix it by manually replacing `window.` to `fft::window::win_type::`.

moatom, 2022-10-15:
About this part, we seems to be able to fix the problem soon.

See https://github.com/gnuradio/gnuradio/blob/main/gr-filter/grc/filter_low_pass_filter.block.yml#L89.
Just change
```
${win.replace('firdes.', 'gr::filter::firdes::')},
```
to
```
${win.replace('window.', 'fft::window::win_type::')},
```
.

marcusmueller, 2024-02-01:
Can still reproduce: https://gist.github.com/marcusmueller/6baaf9667168bdd065c74b553595bd8f