gnuradio/gnuradio #6105: Regression: Blocks with history lose samples on top_block::lock()/top_block::unlock()

vladisslav2011, 2022-08-22:
### What happened?

Any GNU Radio block with history loses some samples on top_block::lock() followed by top_block::unlock().
Problematic commit: 788827ae116bef871e144abd39b1e4482208eabe

### System Information

OS: Linux
GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

3.10.3.0

### Steps to Reproduce the Problem

1. Clone GNU Radio repository
2. Check out main
3. Compile and install
4. Compile test program (cpp and makefile below)
5. Run

main.cpp
```
#include <iostream>
#include <cstdint>
#include <gnuradio/block.h>
#include <gnuradio/sync_block.h>
#include <gnuradio/top_block.h>
#include <gnuradio/io_signature.h>

#if GNURADIO_VER == 38
#define GRSPTR boost::shared_ptr
#elif GNURADIO_VER == 37
#define GRSPTR boost::shared_ptr
#else
#define GRSPTR std::shared_ptr
#endif

class cnt_source : virtual public gr::sync_block
{
public:
    typedef GRSPTR<cnt_source> sptr;
    static sptr make(size_t nitems)
    {
        return gnuradio::get_initial_sptr(new cnt_source(nitems));
    }
    cnt_source(size_t nitems)
    : sync_block("cnt_source",
                 gr::io_signature::make(0, 0, 0),
                 gr::io_signature::make(1, 1, sizeof(uint32_t))),
                 d_cnt(0),
                 d_nitems(nitems)
    {
    }
    virtual ~cnt_source() {};
    int work(int noutput_items,
                            gr_vector_const_void_star& input_items,
                            gr_vector_void_star& output_items)
    {
        uint32_t* o = (uint32_t*)output_items[0];
        if(d_cnt >= d_nitems)
            return -1;
        for (size_t n = 0; n < noutput_items; n++)
        {
            if(d_cnt == d_nitems)
                return n;
            o[n] = d_cnt;
            d_cnt++;
        }
        return noutput_items;
    }
    void reset_counter()
    {
        d_cnt = 0;
    }
    uint32_t get_counter()
    {
        return d_cnt;
    }
private:
    uint32_t d_cnt;
    uint32_t d_nitems;
};

class cnt_sink : virtual public gr::sync_block
{
public:
    typedef GRSPTR<cnt_sink> sptr;
    static sptr make()
    {
        return gnuradio::get_initial_sptr(new cnt_sink());
    }
    cnt_sink()
    : sync_block("cnt_sink",
                 gr::io_signature::make(1, 1, sizeof(uint32_t)),
                 gr::io_signature::make(0, 0, 0)),
                 d_cnt(0)
    {
    }
    virtual ~cnt_sink() {};
    int work(int noutput_items,
                            gr_vector_const_void_star& input_items,
                            gr_vector_void_star& output_items)
    {
        uint32_t* i = (uint32_t*)input_items[0];
        for (size_t n = 0; n < noutput_items; n++)
        {
            d_cnt ++;
        }
        return noutput_items;
    }
    void reset_counter()
    {
        d_cnt = 0;
    }
    uint32_t get_counter()
    {
        return d_cnt;
    }
private:
    uint32_t d_cnt;
};

class hist_block : virtual public gr::sync_block
{
public:
    typedef GRSPTR<hist_block> sptr;
    static sptr make()
    {
        return gnuradio::get_initial_sptr(new hist_block());
    }
    hist_block()
    : sync_block("hist_block",
                 gr::io_signature::make(1, 1, sizeof(uint32_t)),
                 gr::io_signature::make(1, 1, sizeof(uint32_t)))
    {
    }
    virtual ~hist_block() {};
    int work(int noutput_items,
                            gr_vector_const_void_star& input_items,
                            gr_vector_void_star& output_items)
    {
        uint32_t* o = (uint32_t*)output_items[0];
        uint32_t* i = (uint32_t*)input_items[0];
        for (size_t n = 0; n < noutput_items; n++)
        {
            o[n] = i[n+history()];
        }
        return noutput_items;
    }
};

int main()
{
    gr::top_block_sptr tb = gr::make_top_block("top");
    cnt_source::sptr src = cnt_source::make(1000000);
    cnt_sink::sptr dst = cnt_sink::make();
    hist_block::sptr cpy = hist_block::make();
    src->set_output_multiple(1000);
    cpy->set_history(50000);

    tb->connect(src, 0, cpy, 0);
    tb->connect(cpy, 0, dst, 0);
    std::cout <<"No locking:\n";
    tb->start();
    tb->wait();
    std::cout
        <<"Sent: "<<src->get_counter()
        <<" Received: "<<dst->get_counter()
        <<" Lost: "<<(src->get_counter()-dst->get_counter())
        <<"\n";

    src->reset_counter();
    dst->reset_counter();

    std::cout <<"With locking:\n";
    tb->start();
    while(src->get_counter()<10000);
    tb->lock();
    tb->unlock();
    tb->wait();
    std::cout
        <<"Sent: "<<src->get_counter()
        <<" Received: "<<dst->get_counter()
        <<" Lost: "<<(src->get_counter()-dst->get_counter())
        <<"\n";
    return (src->get_counter()-dst->get_counter()) ? 1 : 0;
}
```

Makefile

```
LD_FLAGS=-lgnuradio-runtime -lgnuradio-blocks -lgnuradio-pmt -lfmt -llog4cpp

main: main.cpp
	g++ -DGNURADIO_VER=`gnuradio-config-info --version|sed -e 's/[^0-9]//g'|cut -b1-2` -o main main.cpp $(LD_FLAGS) 
```

### Relevant log output

```shell
Expected (GNU Radio 3.9):

./main 
No locking:
Sent: 1000000 Received: 1000000 Lost: 0
With locking:
Sent: 1000000 Received: 1000000 Lost: 0
```

Observed (Git HEAD at time of issue submission):
```
./main 
No locking:
Sent: 1000000 Received: 1000000 Lost: 0
With locking:
Sent: 1000000 Received: 900001 Lost: 99999
```
```

willcode, 2022-08-22:
This bug report comes from Gqrx, which locks/unlocks whenever a different demod is selected, for example.

vladisslav2011, 2022-08-22:
This regression comes from ~3.8, where hardware source blocks were changed to handle `start`/`stop` differently: really stop (start) the stream and discard both device internal buffers and application side buffers instead of doing nothing. Doing nothing in `start`/`stop` and handling streaming in constructor/destructor was better.
3.7=>3.8 small regression, not addressed by this issue
3.8=>3.9 no change
3.9=>3.10 large regression due to changed `reset_nitem_counters`, addressed by this issue

The best solution to initial problem would be to add a new block method `pause`, for example, to notify a block, that `work`/`general_work` will not be called for some time, and `start`/`stop` will be called soon.
In case of `unlock` pause blocks, instead of stopping, ask the scheduler to not call `work` and wait for threads to finish, create new `flat_flowgraph`, `merge_connections`, reusing buffers/buffer readers, update counters, move tags, and restart.

Gqrx is a good example of interactive c++ GUI application. Playing with it uncovers some interesting bugs related to flowgraph reconfiguration.

vladisslav2011, 2022-08-22:
I'm preparing a PR. What is the best place for the test?
I think, gr-blocks/lib/qa_gr_flowgraph.cc or gr-blocks/lib/qa_gr_top_block.cc may be a good place.
Which one to choose?

willcode, 2022-08-22:
Does anything count on `reset_nitem_counters` being reset? Seems like it would be much easier to just back out that change if not.

vladisslav2011, 2022-08-22:
I agree. It looks, like it would be better to not reset anything as new blocks would have correctly initialized buffer and buffer_readers and old blocks would keep their old buffers having some data and valid read/write pointers. It may be good to not touch tags too. But it requires some testing.

willcode, 2022-08-22:
@mormj @dsorber any opinions on skipping reset of counters on lock/unlock? This was added as part of the custom buffers work.

vladisslav2011, 2022-08-22:
I did more tests with slightly modified test app and found, that tag propagation is broken too. Tags are either dropped or starting to have incorrect offsets in newly connected branches if the line https://github.com/gnuradio/gnuradio/blob/main/gnuradio-runtime/lib/flat_flowgraph.cc#L363 is removed. This should be trivial to fix, but, I think, discussion on this should go to a separate issue.

dsorber, 2022-08-23:
It's not clear to me what `lock`/`unlock` are for (pausing the flowgraph?) but regardless @vladisslav2011's analysis above seems correct.  My understanding is that calls to `start`/`stop` are explicitly expected to clear the state of the flowgraph and therefore the call to `reset_nitem_counters` is needed (without it I think the "inverse" of this problem would exist, or ever stranger problems if the flowgraph was reconfigured).  If `lock`/`unlock` are _not_ expected to clear the state of the flowgraph then they need to be handled differently as was pointed out.

mormj, 2022-08-24:
I will go with @dsorber on this as he has the deepest understanding of the implementation here.

vladisslav2011, 2022-10-04:
Demonstration video (sort of Gqrx, GNU Radio 3.7 ... git HEAD)
https://www.youtube.com/watch?v=82wKYd3FBQs

argilo, 2022-11-11:
This problem doesn't seem to be related to history. The following script demonstrates the problem, and it does not include any blocks with history:
```python
#!/usr/bin/env python3

from gnuradio import gr, blocks

in_data = list(range(100000))

tb = gr.top_block()
src = blocks.vector_source_f(in_data)
thr = blocks.throttle(gr.sizeof_float, 100000)
filt = blocks.copy(gr.sizeof_float)
dst = blocks.vector_sink_f()
tb.connect(src, thr, filt, dst)

tb.start()
while len(dst.data()) == 0:
    pass
tb.lock()
tb.unlock()
tb.wait()
tb.stop()

out_data = list(dst.data())
print(len(in_data), len(out_data))
print(in_data == out_data)
```

On GNU Radio 3.10 & `main`, the output is:
```
100000 83617
False
```
On GNU Radio 3.8 & 3.9, the output is:
```
100000 100000
True
```

argilo, 2022-11-11:
> It's not clear to me what `lock`/`unlock` are for (pausing the flowgraph?)

Here's the documentation:

https://github.com/gnuradio/gnuradio/blob/c11667ef6ebbe528289f528bb303d4668732e942/gnuradio-runtime/include/gnuradio/top_block.h#L86-L95

If you want to modify a flowgraph while it is running (which Gqrx does extensively, for instance to switch between different demodulators), you have to `lock()` the flow graph, then make your changes, then `unlock()`. This allows all changes to occur atomically.

argilo, 2022-11-11:
> This problem doesn't seem to be related to history.

I suppose the connection to history is that a block with history will always have samples in its input buffer, and the input buffer is emptied due to the changes in 788827ae116bef871e144abd39b1e4482208eabe. Blocks with a large history would always lose a large number of samples.

marcusmueller, 2024-02-01:
Can still reproduce [argilo's Python example](https://gist.github.com/marcusmueller/e6119a8da8b3afe5af24276d14363295); so, still an issue