gnuradio/gnuradio #6106: gr-soapy: Add timestamp tag to source output

NZSmartie, 2022-08-22:
### Feature Description

SoapySDR will provide the hardware timestamp of the received buffer in the form of nanoseconds. gr-soapy's source block discards this timetamp `time_ns` when outputting items. 

https://github.com/gnuradio/gnuradio/blob/6c34c6bb795639a7f85ed74421ef1316999da4b8/gr-soapy/lib/source_impl.cc#L51-L97

Can we add a time tag to the output much like gr-uhd does (and possible the frequency and rate as well?)
https://github.com/gnuradio/gnuradio/blob/1dcf8740f4358edee5a6fda7fabd97907817c217/gr-uhd/lib/usrp_source_impl.cc#L623-L630

### Feature Urgency

medium (would be nice to have in the near future)

willcode, 2022-08-22:
Since hardware timestamps are not provided by all hardware, there would need to be a switch for this.

NZSmartie, 2022-08-28:
AFAIK there's no easy way to detect if a Soapy driver's Stream API supports timestamps, it just gets filled out if supported. untouched otherwise. 
the drivers that support timestamps are:

- HackRF - https://github.com/pothosware/SoapyHackRF/blob/79077f0425f732954c1d74dbbacd788b0ac7d733/HackRF_Streaming.cpp#L623
- BladeRF - https://github.com/pothosware/SoapyBladeRF/blob/85f6dc554ed4c618304d99395b19c4e1523675b0/bladeRF_Streaming.cpp#L385
- RTLSDR - https://github.com/pothosware/SoapyRTLSDR/blob/40d0708b009cac1a77f9a7c4052b2f3951b08f60/Streaming.cpp#L509
- LimeSDR - https://github.com/myriadrf/LimeSuite/blob/f81c496176ca7e861d0b0f9a8dcb4cea62b1176d/SoapyLMS7/Streaming.cpp#L447

These Soapy drivers don't populate the timestanp:
- PlutoSDR - https://github.com/pothosware/SoapyPlutoSDR/blob/a07c37230369653818b3a5c448c00cee1ac9f8e5/PlutoSDR_Streaming.cpp#L239
- AirSpyHF - https://github.com/pothosware/SoapyAirspyHF/blob/d6825331a6bfc3578dc6fe5091fce3d550b6af3c/Streaming.cpp#L282

Since each driver supported is defined as a seperate source and sink block in gr-soapy. it might be worth while adding a argument to the `other_settings` parameter.

Alternatively, since each driver that doesn't support hardware timestamps, does not alter its value. Setting it to -1 prior to calling readStream() and checking for any positive value be a requirement for adding the tag.

commit 3eea1057 (2022-09-12): gr-soapy: Add gr-uhd tags to source blocks

Tag the output stream with the frequency, sample rate and hardware time.
Following gr-uhd's rx_freq, rx_rate and rx_time tags.

Closes #6106
PR #6229 (2022-10-08): gr-soapy: Add gr-uhd tags to source blocks (backport to maint-3.10)
Tag the output stream with the frequency, sample rate and hardware time. Following gr-uhd's rx_freq, rx_rate and rx_time tags.

Closes #6106

(cherry picked from commit 3eea1057c53842476f809a62499ee6b8dc080d82)

Backport https://github.com/gnuradio/gnuradio/pull/6117
commit 915601c4 (2022-10-08): gr-soapy: Add gr-uhd tags to source blocks

Tag the output stream with the frequency, sample rate and hardware time.
Following gr-uhd's rx_freq, rx_rate and rx_time tags.

Closes #6106

(cherry picked from commit 3eea1057c53842476f809a62499ee6b8dc080d82)
closed 2022-09-12 by mormj