gnuradio/gnuradio #6110: Tags loss on flowgraph reconfiguration

vladisslav2011, 2022-08-22:
### What happened?

Some tag are lost on flowgraph reconfiguration even on flowgraph branches, that are not changed.
Event `top_block::lock()` followed by `top_block::unlock()` results in tag loss.

### System Information

OS: Linux
GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.10.0.0-rc1-263-g69c5d33b

### Steps to Reproduce the Problem

1. Clone GNU Radio repository
2. Check out main
3. Compile and install
4. Compile test program (cpp and makefile below)
5. Run

main.cpp
```
#include <iostream>
#include <cstdint>
#include <gnuradio/block.h>
#include <gnuradio/sync_block.h>
#include <gnuradio/top_block.h>
#include <gnuradio/io_signature.h>

#if GNURADIO_VER == 38
#define GRSPTR boost::shared_ptr
#elif GNURADIO_VER == 37
#define GRSPTR boost::shared_ptr
#else
#define GRSPTR std::shared_ptr
#endif

class cnt_source : virtual public gr::sync_block
{
public:
    typedef GRSPTR<cnt_source> sptr;
    static sptr make(size_t nitems)
    {
        return gnuradio::get_initial_sptr(new cnt_source(nitems));
    }
    cnt_source(size_t nitems)
    : sync_block("cnt_source",
                 gr::io_signature::make(0, 0, 0),
                 gr::io_signature::make(1, 1, sizeof(uint32_t))),
                 d_nitems(nitems)
    {
    }
    virtual ~cnt_source() {};
    int work(int noutput_items,
                            gr_vector_const_void_star& input_items,
                            gr_vector_void_star& output_items)
    {
        uint32_t* o = (uint32_t*)output_items[0];
        if(d_cnt >= d_nitems)
            return -1;
        for (size_t n = 0; n < noutput_items; n++)
        {
            if(d_cnt == d_nitems)
                return n;
            o[n] = d_cnt;
            if(d_tag_interval && (d_cnt % d_tag_interval == 0))
            {
                add_item_tag(0, nitems_written(0) + n, pmt::intern("cnt_tag"), pmt::from_long(d_cnt));
                d_tags++;
            }
            d_cnt++;
        }
        return noutput_items;
    }
    void reset_counters()
    {
        d_cnt = 0;
        d_tags = 0;
    }
    uint32_t get_counter()
    {
        return d_cnt;
    }
    void set_tag_interval(uint32_t in)
    {
        d_tag_interval = in;
    }
    uint32_t get_tags()
    {
        return d_tags;
    }
private:
    uint32_t d_tag_interval{0};
    uint32_t d_tags{0};
    uint32_t d_cnt{0};
    uint32_t d_nitems;
};

class cnt_sink : virtual public gr::sync_block
{
public:
    typedef GRSPTR<cnt_sink> sptr;
    static sptr make()
    {
        return gnuradio::get_initial_sptr(new cnt_sink());
    }
    cnt_sink()
    : sync_block("cnt_sink",
                 gr::io_signature::make(1, 1, sizeof(uint32_t)),
                 gr::io_signature::make(0, 0, 0))
    {
    }
    virtual ~cnt_sink() {};
    int work(int noutput_items,
                            gr_vector_const_void_star& input_items,
                            gr_vector_void_star& output_items)
    {
        uint32_t* i = (uint32_t*)input_items[0];
        std::vector<gr::tag_t> work_tags;
        get_tags_in_window(work_tags, 0, 0, noutput_items);
        for (const auto& tag : work_tags)
        {
            d_tags++;
            if(i[tag.offset - nitems_read(0) - history()] != pmt::to_long(tag.value))
                d_offset_tags++;
        }
        for (size_t n = 0; n < noutput_items; n++)
        {
            d_cnt ++;
        }
        return noutput_items;
    }
    void reset_counters()
    {
        d_cnt = 0;
        d_tags = 0;
        d_offset_tags = 0;
    }
    uint32_t get_counter()
    {
        return d_cnt;
    }
    uint32_t get_tags()
    {
        return d_tags;
    }
    uint32_t get_offset_tags()
    {
        return d_offset_tags;
    }
private:
    uint32_t d_cnt{0};
    uint32_t d_tags{0};
    uint32_t d_offset_tags{0};
};

class hist_block : virtual public gr::sync_block
{
public:
    typedef GRSPTR<hist_block> sptr;
    static sptr make()
    {
        return gnuradio::get_initial_sptr(new hist_block());
    }
    hist_block()
    : sync_block("hist_block",
                 gr::io_signature::make(1, 1, sizeof(uint32_t)),
                 gr::io_signature::make(1, 1, sizeof(uint32_t)))
    {
    }
    virtual ~hist_block() {};
    int work(int noutput_items,
                            gr_vector_const_void_star& input_items,
                            gr_vector_void_star& output_items)
    {
        uint32_t* o = (uint32_t*)output_items[0];
        uint32_t* i = (uint32_t*)input_items[0];
        for (size_t n = 0; n < noutput_items; n++)
        {
            o[n] = i[n+history()];
        }
        return noutput_items;
    }
};

int main()
{
    gr::top_block_sptr tb = gr::make_top_block("top");
    cnt_source::sptr src = cnt_source::make(1000000);
    cnt_sink::sptr dst0 = cnt_sink::make();
    cnt_sink::sptr dst1 = cnt_sink::make();
    hist_block::sptr cpy = hist_block::make();
    src->set_output_multiple(100);
    src->set_tag_interval(101);
    cpy->set_history(50000);

    tb->connect(src, 0, cpy, 0);
    tb->connect(cpy, 0, dst0, 0);
    std::cout <<"No locking:\n";
    tb->start();
    tb->wait();
    std::cout
        <<"Sent: "<<src->get_counter()<<" items "<<src->get_tags()<<" tags\n"
        <<" Received: "<<dst0->get_counter()<<" items "<<dst0->get_tags()<<" tags\n"
        <<" Lost: "<<(src->get_counter()-dst0->get_counter())<<" items "<<(src->get_tags()-dst0->get_tags())<<" tags\n"
        <<" Offset tags: "<<dst0->get_offset_tags()
        <<"\n";

    src->reset_counters();
    dst0->reset_counters();
    dst1->reset_counters();

    std::cout <<"With locking:\n";
    tb->start();
    while(src->get_counter()<500000);
    tb->lock();
    tb->connect(cpy, 0, dst1, 0);
    tb->unlock();
    tb->wait();
    std::cout
        <<"Sent: "<<src->get_counter()<<" items "<<src->get_tags()<<" tags\n"
        <<" Received: "<<dst0->get_counter()<<" items "<<dst0->get_tags()<<" tags\n"
        <<" Lost: "<<(src->get_counter()-dst0->get_counter())<<" items "<<(src->get_tags()-dst0->get_tags())<<" tags\n"
        <<" Offset tags: "<<dst0->get_offset_tags()
        <<"\n"
        <<"Sink2\n"
        <<" Received: "<<dst1->get_counter()<<" items "<<dst1->get_tags()<<" tags\n"
        <<"Offset tags on sink1: "<<dst1->get_offset_tags()
        <<"\n";
    return (src->get_counter()-dst0->get_counter()) ? 1 : 0;
}
```

Makefile

```
LD_FLAGS=-lgnuradio-runtime -lgnuradio-blocks -lgnuradio-pmt -lfmt -llog4cpp

main: main.cpp
	g++ -DGNURADIO_VER=`gnuradio-config-info --version|sed -e 's/[^0-9]//g'|cut -b1-2` -o main main.cpp $(LD_FLAGS) 
```

### Relevant log output

### Relevant log output

```shell
./main
No locking:
Sent: 1000000 items 9901 tags
 Received: 1000000 items 9901 tags
 Lost: 0 items 0 tags
 Offset tags: 0
With locking:
Sent: 1000000 items 9901 tags
 Received: 924276 items 9151 tags
 Lost: 75724 items 750 tags
 Offset tags: 4344
Sink2
 Received: 438801 items 4344 tags
Offset tags on sink1: 4344
```

vladisslav2011, 2022-08-23:
I have some progress on this. Simple tests are passing.
[x] Adding a sink
[x] Adding a sink, then removing it
[ ] Adding a block with large history
[ ] Adding a block with large alignment
[ ] Removing a block with large history
[ ] Removing a block with large alignment
[ ] Swapping processing block with another having larger history
[ ] Swapping processing block with another having smaller history
[ ] Swapping processing block with another having larger alignment
[ ] Swapping processing block with another having smaller alignment
Which more complex tests may be done?

marcusmueller, 2024-02-01:
I'd say this is part of #6105, and especially of having to decide what we want lock/unlock to do.