gnuradio/gnuradio #6185: Generate/Run fails silently if the GRC flowgraph has not yet been saved

dkozel, 2022-09-21:
### What happened?

Clicking the Generate or Run buttons has no effect and prints/displays no error message if the flowgraph has not been saved yet.

Possible fixes from best to easiest:
* Display a pop-up dialog saying that the flowgraph needs to be saved before Generating and include a button to "Save Flowgraph" and "Cancel". Launch the regular save dialog box and then return to Generating after a successful save.
* Display a pop-up dialog saying that the flowgraph needs to be saved before Generating and have an "OK" button to close.
* Print a console message saying the file needs to be saved before generating.

### System Information

OS: Ubuntu 20.04
GNU Radio 3.10 installed from the releases PPA on September 1st.

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Create a blank/new flowgraph. Press the Generate or Run buttons.

willcode, 2022-09-21:
This appears to be specific to the case where you create a new flowgraph but don't do anything else. Even moving the samp_rate block a little will result in the expected behavior (a save dialog pops up).

The flowgraph should start in "dirty" state to handle this specific case.

dkozel, 2022-09-29:
Wow, what an edge case. Thanks for figuring out the bad initial state

PR #6611 (2023-04-05): grc: prevent Generate/Run silent fails for unsaved GRC flowgraphs

## Description

Before change : Generate/Run button when clicked on a new flow-graph fails silently with no warning or error message.
After change : Clicking Generate/Run button on an unsaved GRC flow-graph prompts you to save the file.
## Related Issue

Fixes #6185 
## Which blocks/areas does this affect?

## Testing Done
closed 2023-04-05 by willcode