gnuradio/gnuradio #6187: GNU radio companion does not start

ottomc2, 2022-09-22:
I installed it recently and I have an icon but the app does not want to start.
a prompt window comes up, few lines appear in it, and everything disappears in few seconds.
Latest revision radioconda 2002.6.24.0
installed successfully

willcode, 2022-09-22:
@ottomc2 Radioconda is here https://github.com/ryanvolz/radioconda. If you think this is a GNU Radio problem, please post the error you are seeing.

ottomc2, 2022-09-22:
Yes, I fetched the version available on that very page:
https://github.com/ryanvolz/radioconda
There is the only version for windows available there:
radioconda-Windows-x86_64.exe
<https://glare.now.sh/ryanvolz/radioconda/radioconda-.*-Windows-x86_64.exe>
I ran that one and got a message that GNU radio was installed successfully.
After that, I have not been able to start the application.
That's the problem I experience.

willcode, 2022-09-22:
Please move the issue to the radioconda issue tracker. This doesn't look like a GNU Radio issue. You can also try chat.gnuradio.org for help.

ottomc2, 2022-09-22:
ok i copied my message to the radioconda issue tracker

closed 2022-09-22 by ottomc2