gnuradio/gnuradio #6190: Unify network address fields and make them time-varying

nils-werner, 2022-09-26:
### Feature Description

It would be really great if all address fields of all network blocks could follow the same format, and be time-varying.

Right now, the blocks are using different formats:

use two separate fields for IP address `127.0.0.1` and port `2000`

 - TCP Sink
 - UDP Sink

Use one URL field, like `tcp://127.0.0.1:50000`

 - ZMQ PUB Sink
 - ZMQ PUB Message Sink
 - ZMQ PUSH Sink
 - ZMQ PUSH Message Sink
 - ZMQ REP Sink
 - ZMQ REP Message Sink

It would be great if the TCP Sink and UDP Sink could also be migrated over to the URL format.

Secondly, it would be great if these fields could be made time-varying/receive setters. This would then allow users to 

- change endpoints mid-runtine, e.g. in a GUI or from received data
- discover endpoints automatically, e.g. using Zeroconf

### Feature Urgency

medium (would be nice to have in the near future)

dkozel, 2022-09-26:
Unifying the user interface would be very simple and makes sense. Using the IP:port# format does make it less obvious that a port can be set (I wonder what the default is if one is not set), but making sure the documentation is updated and clear would largely address that.

Making them time varying is likely a somewhat larger development task to make sure the connections are shutdown and transitioned cleanly, but certainly nothing that is unusual for networking code. Are you looking at making these changes to support your needs? I see no reason we'd not want to merge those.

willcode, 2022-09-26:
We could also make the GRC yml for ZMQ split out the components of the URI, which would help people who don't know how to make a ZMQ URI. I think I'd lean more in that direction. The underlying calls would remain as they are today.

nils-werner, 2022-09-26:
> Are you looking at making these changes to support your needs?

I don't think I personally will be able to, as I will be leaving our internal GNUradio-related project soon and don't expect to do anything similar in the near time. But I will open a ticket asking for somebody else to have a look at this. Especially combined with ZeroConf this is a big convenience feature for us, so hopefully somebody will find the time.

> We could also make the GRC yml for ZMQ split out the components of the URI, which would help people who don't know how to make a ZMQ URI.

I would suggest the opposite. URIs that contain a port number are not such a difficult concept, and the syntax is used absolutely everywhere. And splitting address and port makes it harder to populate them in a variable (time-varying or not).