gnuradio/gnuradio #6196: Android/runtime: weakptr failures

marcusmueller, 2022-09-29:
### What happened?

As shown in Adrian's presentation at Grcon, in Android hier Block building failed. He says he has a branch that uses a search and replace op to fix the build, and he also mentions that @bastibl has Android-specific fixes in his work that should end up in mainline. 

Investigate.

### System Information

OS: Android
GR Installation Method: Adrian's build mechanism

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Build

adisuciu, 2022-09-29:
In the presentation I mentioned that the crash only really happened on Android, on python, and only when I was using hierarchical blocks in this configuration. The blocks get basically immediatly destroyed after being created if they are part of a hier block. If they are part of the top block, things don't crash. I briefly tried to reproduce this on PC, by logging the block lifecycle, but it didn't reproduce. I didn't give it too much try though ...

@marcusmueller 
https://github.com/analogdevicesinc/gnuradio/commit/a11408eee742e300db00db0d0a7cf1202455b6e5 
https://github.com/analogdevicesinc/gnuradio/commit/3c35dfcad37941d06089325bd5757224037b90bf
this is the commit that adds compatibility to the smart_holder branch. You need to use this version of pybind -  https://github.com/pybind/pybind11/tree/smart_holder 
The port is not full - there is an error when compiling gr-dtv(I think) or gr-digital, but that should be easy to fix .. 

Also these are the commits that I'm talking about that maybe shoulde be mainlined - 
https://github.com/bastibl/gnuradio/commit/4664385d89efecb1c58615e9ad30cdca125c3940
https://github.com/analogdevicesinc/gnuradio/commit/624e8aa23a335fbfa20fedfbb28b6dd6ec29eaa2
@bastibl do you agree ?

-Adrian

bastibl, 2022-09-30:
Regarding thread affinity, I'm not sure, if this is the nicest way to do it. Maybe one could just not call these functions or have an empty implementation on Android?

The Android shared memory implementation was mainly a test to see, if it provides a benefit over the buffer implementation using a mmaped temp file. But I  didn't see any differences in terms of performance. So I would just use the temp file implementation on Android to avoid unnecessary complexity.