gnuradio/gnuradio #6202: VSCode Extension for GR_Modtool/GR

joshjab, 2022-09-30:
### Feature Description

Just creating this issue to store the conversation from the online chat and any mockups.

It'd be nice to have an extension for VSCode to facilitate a workflow for OOT. Something akin to [this](https://marketplace.visualstudio.com/items?itemName=ivannovak1004.gnuradio-integration) plugin that's already on the VSCode marketplace, but with the ability to create new modules, blocks, etc. 

The CLI is perfectly functional as is, but for a new user trying to create a new OOT module and learn the basic workflow, it'd be nice to have a basic 'wizard' somewhere to guide the developer through the first time.

### Feature Urgency

low (just an idea)

AsriFox, 2023-12-03:
I've created this: https://github.com/AsriFox/gnuradio-integration
It's not on a marketplace yet, but you can download VSIX from releases and install it.
It's basically a thin wrapper around the CLI, and some features are not translated to commands. My plan is to create a sidebar tree, that will allow for structured project management.

AsriFox, 2024-01-29:
Update: in addition to modtool functionality through the command palette, a tree view was added that displays blocks in the OOT module. The extension should be compatible with custom installation prefixes and conda environments.

The extension has been released to Open VSIX: [gnuradio-integration](https://open-vsx.org/extension/AsriFox/gnuradio-integration). If you're using VSCodium, it's available from the search; you can also download the VSIX package from [releases](https://github.com/AsriFox/vscode-gnuradio-integration/releases) and install it manually.

For marketplace users: version 0.3.3 ([here](https://marketplace.visualstudio.com/items?itemName=ivannovak1004.gnuradio-integration)) is broken; the 0.3.4 update is not there yet, please install it from other sources.

willcode, 2024-01-29:
Looks easier than using the CLI. Is 0.3.4 going to be on the marketplace soon?

AsriFox, 2024-01-30:
Hopefully. Version 0.3.4 PR is already merged.  
Releases on the marketplace are handled by @IvanNovak1004 for now, but they said they are planning to transfer ownership.

AsriFox, 2024-02-01:
Version 0.3.4 is now released on the [marketplace](https://marketplace.visualstudio.com/items?itemName=ivannovak1004.gnuradio-integration)! You should be able to install it from VSCode.   
Would love to hear your feedback.

AsriFox, 2024-09-04:
Added instructions to the wiki: https://wiki.gnuradio.org/index.php?title=UsingVSCode

willcode, 2024-09-09:
Closing, since AsriFox has support and it's linked from the wiki.

closed 2024-09-09 by willcode