gnuradio/gnuradio #6227: CI OSX Conda build is flaky

argilo, 2022-10-07:
The OSX Conda build sometimes fails. Here are some examples of recent failures:

https://github.com/gnuradio/gnuradio/actions/runs/2986963438/jobs/4812546038
https://github.com/gnuradio/gnuradio/actions/runs/2996076013/jobs/4806418288
https://github.com/gnuradio/gnuradio/actions/runs/2884399579/jobs/4583929663

Just before failing, the following is logged:

```
Partially parsed some '.tbd' files in sysroot /Applications/Xcode_13.2.1.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk, pretending .tbds are their install-names
Adding support to 'conda-build' for parsing these in 'liefldd.py' would be easy and useful:
['usr/lib/swift/libswiftSwiftOnoneSupport.tbd', 'System/Library/PrivateFrameworks/RunningBoardServices.framework/RunningBoardServices.tbd'] ...
WARNING:conda_build.post:Partially parsed some '.tbd' files in sysroot /Applications/Xcode_13.2.1.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk, pretending .tbds are their install-names
Adding support to 'conda-build' for parsing these in 'liefldd.py' would be easy and useful:
['usr/lib/swift/libswiftSwiftOnoneSupport.tbd', 'System/Library/PrivateFrameworks/RunningBoardServices.framework/RunningBoardServices.tbd'] ...
/Users/runner/miniforge3/etc/profile.d/conda.sh: line 9:  5271 Segmentation fault: 11  "$CONDA_EXE" $_CE_M $_CE_CONDA "$@"
/Users/runner/miniforge3/lib/python3.10/multiprocessing/resource_tracker.py:224: UserWarning: resource_tracker: There appear to be 1 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '
Error: Process completed with exit code 139.
```

The specific `.tbd` file being processed at the time differs, but the segmentation fault remains the same.

Unfortunately I don't know enough about Conda to diagnose this further.

ryanvolz, 2022-10-11:
This has been seen in conda-forge builds too, so it's not anything specific to GR. I think it was due to a bad release of `lief`, which has hopefully been fixed since the last release seen here: https://github.com/conda-forge/lief-feedstock/pull/34. At least I haven't seen it crop up in a little while and I don't think I've seen any discussion of it recently.

argilo, 2022-10-11:
Thanks for the insight, @ryanvolz. The most recent instance I found was on September 6. Since it's probably fixed, I'll close this off, and re-open if any new failures occur.

argilo, 2022-10-15:
Re-opening, since a failure occurred today: https://github.com/argilo/gnuradio/actions/runs/3256190531/jobs/5346320765

ryanvolz, 2022-10-17:
> Re-opening, since a failure occurred today: https://github.com/argilo/gnuradio/actions/runs/3256190531/jobs/5346320765

I'll keep an eye out in the conda-forge world to see what the fix will be.

argilo, 2022-10-31:
Still failing as of 2022-10-31: https://github.com/gnuradio/gnuradio/actions/runs/3361223280/jobs/5571354150

ryanvolz, 2022-11-01:
There should be a workaround for this out in the wild soon: https://github.com/conda-forge/conda-smithy/pull/1684.

argilo, 2022-11-01:
I see a new release is out now: https://github.com/conda-forge/conda-smithy/releases/tag/v3.21.3

PR #6332 (2022-11-03): CI: conda: Update recipe and re-render

## Description

This updates the dependencies in the conda recipe to remove `libcblas` following #5913, add `pytest` for running grc_tests, and synchronize with the conda-forge recipe. It also re-renders the conda CI scripts to build against newer dependency versions and fix #6227.

## Related Issue

Fixes #6227

## Which blocks/areas does this affect?

CI

## Testing Done
closed 2022-11-03 by mormj