gnuradio/gnuradio #6252: GR 4.0: standardise synchronisation of data streams from different flow-graphs & nodes using tags

RalphSteinhagen, 2022-10-15:
### Feature Description

This feature request does not require changes to the GR core but proposes standardised conventions for the naming and layout of PMT-based trigger tags and meta-information to facilitate synchronising signal streams across different flow-graphs, nodes, and in- and OOT-source devices that are part of the greater GNU Radio eco-systems. The implementation of the specific synchronisation (blocks) would be initially OOT.

It tackles the real-world use-case in MIMO systems where -- for a large number or physically-distributed signal sources -- not all RX signals can be synchronised by using a single SDR but that require a large and/or distributed DAQ infrastructure.

[image]

This is not a new concept (i.e. already used in some LimeSDR and UHD devices) but merely aims to standardise the existing solutions as vendor-neutral conventions. In the proposed scheme, signal sources should
1. map their external (usually HW-based) or internal(usually SW-based) synchronisation triggers as stream tags that are placed on the sample of the exact trigger occurrence. As a minimum key-value-pair information (`TRIGGER` prefix tbd.):
    * `TRIGGER_NAME` - a 'std::string' (default, e.g. 'PPS' -- to be overwritten by info in 'b)')
    * `TRIGGER_TIME` - a 'uint64_t' UTC- or TAI-based time-stamp in nanoseconds (choice: UTC vs. TAI tbd.)
    * `TRIGGER_OFFSET` - floating-point delay of the sample w.r.t. the trigger in seconds (e.g.compensating analog group delays)
2. provide a 'timing' msg port that receives PMT messages sent ahead of the trigger to
    * change the trigger name, precise time stamp, and offset of the upcoming trigger, for example, based on information from an external high-resolution timing receiver (GPS, White-Rabbit, SW-timing system, ...)
    * add custom meta-information on top of the same trigger tag that is use-case or site-specific (<-> user-extensions)

The main goal is to publicly document/implement this for all in-tree and perhaps also as a recommendation for all OOT signal source blocks to allow participating in and sharing of synchronisation schemes. Blocks not participating in these schemes would be unaffected and merely forward the tag information untouched.

The aspect 'b)' could also be (configurably) used for SW-based synchronisation (e.g. using PTP/NTP) by adding trigger tags whenever the 'timing' msg port is updated without having a physical HW trigger. This would allow hobbyist and academic users without access to high-resolution ns-level timing system (e.g. using an RTL dongle) to experiment and test flow-graphs before they are transferred to operational environments that have these timing systems.

### Feature Urgency

medium (would be nice to have in the near future)

### More Information

The actual implementation of the specific 'Sync' block(s) should initially be OOT until sufficiently generic or widely adopted solutions emerge.

This also, since there are different schemes based on the specific use-case and facility boundary conditions, that favour synchronisation mechanisms deploying
  * zero-padding
  * blocking down-stream flow-graphs until the streams are in-sync
  * Zero-Order-Holder (ZOH)
  * failure mitigation/compensation of:
    * reconnecting/restarting SDRs/nodes
    * no data reception and/or time-outs (e.g. for signals transmitted via non-deterministic networks)
    * clock-drifts
    * transmission delays
    * ...

Open for comments, suggestions, and additional thoughts.

michael-roe, 2022-10-15:
That's an interesting idea.

one possible source for the sync information is the correlate access code tag block.

suppose that you have SDR recordings from physically distant receivers that have loosely synchronised clocks. E.g. you have several ground stations in different locations listening to the same satellite.

you could look for the sync word in each recording, align them on the sync word (turning a loosely synchronised clock into a much more accurately synchronised clock) and then do MIMO.

=====

There's a really horrible trick where you analogue mix a gps pulse per second into the RF input of your sdr, then use software on the iq samples to extract the timing information. Can be used with cheap sides that don't have a proper gps pps input.

this is similar to the previous case, except your sync marker is once per second gps time, rather than once per frame sent over the satellite downlink.

jacobagilbert, 2022-11-01:
As discussed in chat, in general the concept of true wall clock time (or differential WCT) should be avoided wherever possible within DSP, and imparted at edges only when absolutely required. There are many reasons for this, notably the two clock problem, the fact that GR generally does not run on real time operating systems, the need for simulation, offline processing, non-realtime processing, etc, etc, but most importantly because except for rare situations it is not at all important and the inherent 'perfect' sample alignment GR provides as data moves through the flowgraph is a core contract of the framework (and important to the backpressure concept).

There are a few situations where differential/actual WCT is helpful, generally related to interfacing with time-synchronized sources/sinks, benchmarking, or when something must periodically be checked (rare for non-source/sink GR blocks... usually this is best done using sample alignment).