gnuradio/gnuradio #6269: Frequency error in

cptspacemanspiff, 2022-10-19:
### What happened?

fxpt.h is not accurate enough, and does not have enough bits in the lookup index to represent high sample rate signals, this can be seen when trying to generate a 10MHz sinusoid at 25MHz sample rate. 

The cause of this is that:

1. The frequency passed to the accumulator is passed as a float. this limits the accuracy to ~24 bits.
2. The mapping from index to lookup table is using a wrapping uint32_t, this means that you only have ~4 billion samples per cycle. this means that if your step size (for the frequency rotation) is rounded to the nearest uint32 index when it was really halfway between 2 indexes, at 25Mhz within ~160 seconds you will accumulate an additional cycle. 

I implemented a fix where a uint64 is used for the index instead, alongside setting the frequency with a double.

It seems to be working appropriately (I am testing phase synchronization generating a 10Mhz reference with USRP at 25Mhz, and it no longer drifts.)

Attached are file changes where I implemented a fix for this at least the fxpt_nco, though there seem to be a few other files that rely on this implementation.

[Gnuradio_fxpt_fix.zip](https://github.com/gnuradio/gnuradio/files/9816056/Gnuradio_fxpt_fix.zip)

### System Information

OS: Centos8
GR Installation Method: built from source.

### GNU Radio Version

3.10 (maint-3.10)

### Steps to Reproduce the Problem

Generate a high frequency sinusoid using the signal generator block at a high sample rate. The generated output is incorrect.

willcode, 2022-10-19:
Thanks for the report. I can see why this is an issue for precise work at high rates. We need to think of a fix that doesn't change the API, though, if this is going to make it into a 3.10 release. Something like this could make it into 4.0, or 3.11 if we end up having one.

We generally take fixes in the form of Github PRs. If you are not comfortable with git, and want to contribute a patch, please put in the form of `diff` output, and someone else could then easily turn it into a PR.

marcusmueller, 2024-02-01:
highly correlated with #6800.

Regarding the fix: I've been working on this, and note that libc-native `sin` implementations are far from perfect; I do concur with the larger int type reduces the frequency error. But in the end, this is just a 0.025 ppb error, and it does come with a relatively high performance cost (not as badly as having a significantly larger table, but still). It's time to re-evaluate this, let's bundle these efforts on 6800!

closed 2024-02-01 by marcusmueller

ValeZAA, 2024-02-09:
>note that libc-native sin implementations are far from perfect;

They are garbage, yes. Glibc code for sincos is horrible, see first page of this https://members.loria.fr/PZimmermann/talks/core-math-fpbench.pdf

And also code for that is on gitlab, linked from here https://core-math.gitlabpages.inria.fr/