gnuradio/gnuradio #6277: close() function in File Source & File Sink doesn't close file

argilo, 2022-10-22:
While working on #6275 and #6276, I noticed that calling `close()` on a File Sink block does not actually close the open file. This is particularly problematic on Windows, where it is not possible to delete a file until it is closed. The `close()` function closes `d_new_fp`, but the currently active file is in fact `d_fp`:

https://github.com/gnuradio/gnuradio/blob/962eb04aaef345bcdc808d98cd4621189237539a/gr-blocks/lib/file_sink_base.cc#L95-L105

The File Source block shares the same bug:

https://github.com/gnuradio/gnuradio/blob/962eb04aaef345bcdc808d98cd4621189237539a/gr-blocks/lib/file_source_impl.cc#L236-L246

In #6275 and #6276, I worked around this by restructuring tests to delete files only after the entire flowgraph is destroyed. For other applications, this approach might not be practical.

willcode, 2022-10-22:
The semantics of `close()` are not obvious here. `open()` specifies the _next_ file to "play", and `close()` closes the _next_ file if there is one. Whether that interpretation of `close()` makes it useful, I'm not sure.

However, closing the current file could mean (1) immediately stop plaing the current file and switch to the next one if any, (2) close the current (and potentially next) file and signal DONE since the block would be out of source items (or space for sink items).

Since the documentation does not make clear what `close()` means, let's decide on that first then put in a fix.

argilo, 2022-10-22:
For reference, here's what the docs say:

https://github.com/gnuradio/gnuradio/blob/962eb04aaef345bcdc808d98cd4621189237539a/gr-blocks/include/gnuradio/blocks/file_source.h#L65-L79

https://github.com/gnuradio/gnuradio/blob/962eb04aaef345bcdc808d98cd4621189237539a/gr-blocks/include/gnuradio/blocks/file_sink_base.h#L45-L56

argilo, 2022-10-22:
> open() specifies the next file to "play", and close() closes the next file if there is one.

The `work()` function calls `do_update()` on every invocation, and `do_update()` switches over to the new file, if there is one. So the transition to a new file will be more or less immediate, unless the flow graph is stopped. That makes me think that option (2) was probably intended.

willcode, 2022-10-22:
`file_sink` blackholes items while there is no open file, and does not terminate the flowgraph on `close()`. Then, to be symmetric, `file_source` should just block when there is no open file. In that state, though, nothing would terminate a typical flowgraph. So, I guess (2) seems less likely to cause problems, but is not symmetric with `file_sink` and it seems like users should not need to explicitly call `close()`.

There are other blocks (SDR, ZMQ, etc.) that have open resources and we don't make the user explicitly close them, so I wouldn't want to start a precedent with `file_source`.

I think the original intent of `file_source::close()` was to close an "on deck" file. I was going to claim credit for the confusion, but it looks like I just fixed indentation on those lines in 2018 while adding other features.

argilo, 2022-10-22:
> There are other blocks (SDR, ZMQ, etc.) that have open resources and we don't make the user explicitly close them, so I wouldn't want to start a precedent with file_source.

True. If the File blocks closed their files automatically when the flowgraph stopped executing, then the tests I was working on would have worked just fine, no `close()` or deallocation required.

willcode, 2022-10-22:
Should files be closed on `stop()` and the re/opened on `start()`? That would make sense, and other sources/sinks do similar things. Currently `file_sink` does a flush on `stop()` but does not close the file. It does nothing on `start()`. Would it need to seek/append on `start()`?. `file_source` does nothing on `stop()` or `start()`.

The file_descriptor, file_meta, tagged_file, and wavefile blocks would also need some investigation.

argilo, 2022-10-22:
> Should files be closed on `stop()` and the re/opened on `start()`?

I think it would make sense to do that.

> Would it need to seek/append on `start()`?

Yes, I think that would be the right behaviour. If you `stop()` a flowgraph to reconfigure it, you'd probably expect existing file sources & sinks to pick up where they left off when you `start()`.

marcusmueller, 2024-02-01:
I don't see us having gotten around to fixing this, yet.