gnuradio/gnuradio #6278: gr 3.10.4 fails to build with recent spdlog/fmt versions

mnhauke, 2022-10-22:
### What happened?

GR 3.10 fails to build with a modern toolchain (gcc 12.2.1, fmt 9.1.0, spdlog 1.10.0)
```
[  251s] /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc:114:22:   required from here
[  251s] /usr/include/fmt/core.h:1757:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
[  251s]  1757 |       formattable,
[  251s]       |       ^~~~~~~~~~~
[  251s] /usr/include/fmt/core.h:1757:7: note: 'formattable' evaluates to false
```

### System Information

OS: (openSUSE Tumbleweed)
GR Installation Method: (manual compilation)
gcc: (SUSE Linux) 12.2.1 20220830 [revision e927d1cf141f221c5a32574bde0913307e140984]
fmt: 9.1.0
spdlog: 1.10.0 (with SPDLOG_FMT_EXTERNAL=ON)

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.4

### Steps to Reproduce the Problem

Try to build it on openSUSE Tumbleweed

full buildlog:
https://build.opensuse.org/package/live_build_log/hardware:sdr/gnuradio/openSUSE_Tumbleweed/x86_64

### Relevant log output

```shell
[  251s] FAILED: gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/controlport/thrift/rpcpmtconverters_thrift.cc.o 
[  251s] /usr/bin/c++ -DBOOST_DLL_USE_STD_FS -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DFMT_SHARED -DGR_CTRLPORT -DGR_MPLIB_GMP -DGR_PERFORMANCE_COUNTERS -DHAVE_COSF -DHAVE_GETPAGESIZE -DHAVE_LIBUNWIND -DHAVE_MMAP -DHAVE_PTHREAD_SETSCHEDPARAM -DHAVE_PTHREAD_SIGMASK -DHAVE_SCHED_SETSCHEDULER -DHAVE_SHM_OPEN -DHAVE_SIGACTION -DHAVE_SIGNAL_H -DHAVE_SINCOS -DHAVE_SINCOSF -DHAVE_SINF -DHAVE_SYSCONF -DHAVE_SYS_IPC_H -DHAVE_SYS_MMAN_H -DHAVE_SYS_RESOURCE_H -DHAVE_SYS_SHM_H -DHAVE_SYS_TYPES_H -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -DTRY_SHM_VMCIRCBUF -Dgnuradio_runtime_EXPORTS -I/home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/build/gnuradio-runtime/lib/math -I/home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/../include -I/home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/build/gnuradio-runtime/lib/../include -I/home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib -I/home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/build/gnuradio-runtime/lib/controlport -I/home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/pmt/../../include -isystem /usr/include/python3.10 -O2 -Wall -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=3 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -fvisibility=hidden -Wsign-compare -Wno-uninitialized -fcx-limited-range -O2 -g -DNDEBUG -fPIC -std=c++17 -Winvalid-pch -include /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/build/gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/cmake_pch.hxx -MD -MT gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/controlport/thrift/rpcpmtconverters_thrift.cc.o -MF gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/controlport/thrift/rpcpmtconverters_thrift.cc.o.d -o gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/controlport/thrift/rpcpmtconverters_thrift.cc.o -c /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc
[  251s] In file included from /usr/include/spdlog/fmt/fmt.h:27,
[  251s]                  from /usr/include/spdlog/common.h:45,
[  251s]                  from /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/../include/gnuradio/logger.h:36,
[  251s]                  from /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/../include/gnuradio/custom_lock.h:15,
[  251s]                  from /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/../include/gnuradio/buffer.h:15,
[  251s]                  from /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/../include/gnuradio/buffer_double_mapped.h:15,
[  251s]                  from /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/../include/gnuradio/io_signature.h:17,
[  251s]                  from /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/../include/gnuradio/basic_block.h:15,
[  251s]                  from /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/../include/gnuradio/block.h:17,
[  251s]                  from /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/../include/gnuradio/sync_block.h:15,
[  251s]                  from /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/build/gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/cmake_pch.hxx:5,
[  251s]                  from <command-line>:
[  251s] /usr/include/fmt/core.h: In instantiation of 'constexpr fmt::v9::detail::value<Context> fmt::v9::detail::make_value(T&&) [with Context = fmt::v9::basic_format_context<fmt::v9::appender, char>; T = const std::shared_ptr<pmt::pmt_base>&]':
[  251s] /usr/include/fmt/core.h:1777:29:   required from 'constexpr fmt::v9::detail::value<Context> fmt::v9::detail::make_arg(T&&) [with bool IS_PACKED = true; Context = fmt::v9::basic_format_context<fmt::v9::appender, char>; type <anonymous> = fmt::v9::detail::type::custom_type; T = const std::shared_ptr<pmt::pmt_base>&; typename std::enable_if<IS_PACKED, int>::type <anonymous> = 0]'
[  251s] /usr/include/fmt/core.h:1901:77:   required from 'constexpr fmt::v9::format_arg_store<Context, Args>::format_arg_store(T&& ...) [with T = {const std::shared_ptr<pmt::pmt_base>&}; Context = fmt::v9::basic_format_context<fmt::v9::appender, char>; Args = {std::shared_ptr<pmt::pmt_base>}]'
[  251s] /usr/include/fmt/core.h:1918:31:   required from 'constexpr fmt::v9::format_arg_store<Context, typename std::remove_cv<typename std::remove_reference<Args>::type>::type ...> fmt::v9::make_format_args(Args&& ...) [with Context = basic_format_context<appender, char>; Args = {const std::shared_ptr<pmt::pmt_base>&}]'
[  251s] /usr/include/spdlog/logger.h:370:68:   required from 'void spdlog::logger::log_(spdlog::source_loc, spdlog::level::level_enum, spdlog::string_view_t, Args&& ...) [with Args = {const std::shared_ptr<pmt::pmt_base>&}; spdlog::string_view_t = fmt::v9::basic_string_view<char>]'
[  251s] /usr/include/spdlog/logger.h:90:13:   required from 'void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, fmt::v9::format_string<T ...>, Args&& ...) [with Args = {const std::shared_ptr<pmt::pmt_base>&}; fmt::v9::format_string<T ...> = fmt::v9::basic_format_string<char, const std::shared_ptr<pmt::pmt_base>&>]'
[  251s] /usr/include/spdlog/logger.h:96:12:   required from 'void spdlog::logger::log(spdlog::level::level_enum, fmt::v9::format_string<T ...>, Args&& ...) [with Args = {const std::shared_ptr<pmt::pmt_base>&}; fmt::v9::format_string<T ...> = fmt::v9::basic_format_string<char, const std::shared_ptr<pmt::pmt_base>&>]'
[  251s] /usr/include/spdlog/logger.h:170:12:   required from 'void spdlog::logger::error(fmt::v9::format_string<T ...>, Args&& ...) [with Args = {const std::shared_ptr<pmt::pmt_base>&}; fmt::v9::format_string<T ...> = fmt::v9::basic_format_string<char, const std::shared_ptr<pmt::pmt_base>&>]'
[  251s] /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/../include/gnuradio/logger.h:177:24:   required from 'void gr::logger::error(const spdlog::string_view_t&, const Args& ...) [with Args = {std::shared_ptr<pmt::pmt_base>}; spdlog::string_view_t = fmt::v9::basic_string_view<char>]'
[  251s] /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc:114:22:   required from here
[  251s] /usr/include/fmt/core.h:1757:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
[  251s]  1757 |       formattable,
[  251s]       |       ^~~~~~~~~~~
[  251s] /usr/include/fmt/core.h:1757:7: note: 'formattable' evaluates to false
```

willcode, 2022-10-22:
Should `knob` should be `knob.type` here https://github.com/gnuradio/gnuradio/blob/962eb04aaef345bcdc808d98cd4621189237539a/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc#L114 to match https://github.com/gnuradio/gnuradio/blob/962eb04aaef345bcdc808d98cd4621189237539a/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc#L257?

mnhauke, 2022-10-22:
changing this
```
--- a/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc
+++ b/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc
@@ -111,7 +111,7 @@ GNURadio::Knob rpcpmtconverter::from_pmt(const pmt::pmt_t& knob)
         // FIXME: Don't get loggers every time we need to log something.
         gr::logger_ptr logger, debug_logger;
         gr::configure_default_loggers(logger, debug_logger, "rpcpmtconverter");
-        logger->error("ERROR Don't know how to handle Knob Type (from): {}", knob);
+        logger->error("ERROR Don't know how to handle Knob Type (from): {}", knob.type);
         assert(0);
     }
     return GNURadio::Knob();
```

results in another error:
```
[   79s] /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc: In function 'GNURadio::Knob rpcpmtconverter::from_pmt(const pmt::pmt_t&)':
[   79s] /home/abuild/rpmbuild/BUILD/gnuradio-3.10.4.0/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc:114:83: error: 'const pmt::pmt_t' {aka 'const class std::shared_ptr<pmt::pmt_base>'} has no member named 'type'
[   79s]   114 |         logger->error("ERROR Don't know how to handle Knob Type (from): {}", knob.type);
[   79s]       |                                                                                   ^~~~
```

willcode, 2022-10-22:
Oh, different types. Might just need pmt::write_string() where an implicit cast used to work. Not at a computer right now so will look later.

willcode, 2022-10-23:
See if this works for you. It compiles here and I think it does what is intended. I don't have fmt v9 installed. @marcusmueller do you have any comments here?
```
diff --git a/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc b/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc
index 8ef3db1fe..2e8917d51 100644
--- a/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc
+++ b/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc
@@ -111,7 +111,7 @@ GNURadio::Knob rpcpmtconverter::from_pmt(const pmt::pmt_t& knob)
         // FIXME: Don't get loggers every time we need to log something.
         gr::logger_ptr logger, debug_logger;
         gr::configure_default_loggers(logger, debug_logger, "rpcpmtconverter");
-        logger->error("ERROR Don't know how to handle Knob Type (from): {}", knob);
+        logger->error("ERROR Don't know how to handle Knob Type (from): {}", pmt::write_string(knob));
         assert(0);
     }
     return GNURadio::Knob();
```

mnhauke, 2022-10-23:
Thanks, this is working fine here.

PR #6288 (2022-10-31): runtime: make logging in rpcpmtconverters_thift work with fmt v9
## Description
Implicit cast in an error/log message did not work with newer versions of fmt (v9 on SUSE). Use `pmt::write_string()` instead.

## Related Issue
Fixes #6278

## Which blocks/areas does this affect?
Runtime

## Testing Done

## Checklist
- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [x] I have squashed my commits to have one significant change per commit. 
- [x] I [have signed my commits before making this PR](https://github.com/gnuradio/gnuradio/blo [... cut]
closed 2022-10-31 by mormj