gnuradio/gnuradio #6283: Segfault on macOS when running a flowgraph containing a Python block

argilo, 2022-10-23:
Many CI tests segfault on macOS, and thus need to be excluded from running in Conda builds. I looked into a few of the failures, and it appears that a segfault occurs whenever a flowgraph containing a Python block is started. The following script demonstrates a crash:
```python
import numpy as np
from gnuradio import gr, blocks

class foo_cc(gr.sync_block):
    def __init__(self):
        gr.sync_block.__init__(
            self,
            name='foo',
            in_sig=[np.complex64],
            out_sig=[np.complex64]
        )

    def work(self, input_items, output_items):
        output_items[0][:] = input_items[0]
        return len(output_items[0])

tb = gr.top_block()

src = blocks.vector_source_c([1, 2, 3])
foo = foo_cc()
dst = blocks.vector_sink_c()

tb.connect(src, foo)
tb.connect(foo, dst)
tb.run()
```
Backtrace:
```
* thread #15, stop reason = EXC_BAD_ACCESS (code=1, address=0x1b0)
  * frame #0: 0x00007ff801b20882 libsystem_pthread.dylib`pthread_mutex_lock + 4
    frame #1: 0x0000000106a57670 Python`take_gil + 76
    frame #2: 0x0000000106ab0270 Python`PyGILState_Ensure + 48
    frame #3: 0x0000000105bab61c libgnuradio-runtime.v3.11.0.0git-257-g83ba4690.dylib`pybind11::detail::get_internals() + 60
    frame #4: 0x0000000105bab55c libgnuradio-runtime.v3.11.0.0git-257-g83ba4690.dylib`pybind11::gil_scoped_acquire::gil_scoped_acquire() + 28
    frame #5: 0x0000000105baabc3 libgnuradio-runtime.v3.11.0.0git-257-g83ba4690.dylib`gr::block_gateway_impl::start() + 35
    frame #6: 0x0000000105b4eb62 libgnuradio-runtime.v3.11.0.0git-257-g83ba4690.dylib`gr::block_executor::block_executor(std::__1::shared_ptr<gr::block>, int) + 322
    frame #7: 0x0000000105ba0847 libgnuradio-runtime.v3.11.0.0git-257-g83ba4690.dylib`gr::tpb_thread_body::tpb_thread_body(std::__1::shared_ptr<gr::block>, std::__1::shared_ptr<boost::barrier>, int) + 87
    frame #8: 0x0000000105b95c68 libgnuradio-runtime.v3.11.0.0git-257-g83ba4690.dylib`gr::tpb_container::operator()() + 88
    frame #9: 0x0000000105b95af3 libgnuradio-runtime.v3.11.0.0git-257-g83ba4690.dylib`gr::thread::thread_body_wrapper<gr::tpb_container>::operator()() + 35
    frame #10: 0x000000010518eba2 libboost_thread-mt.dylib`boost::(anonymous namespace)::thread_proxy(void*) + 146
    frame #11: 0x00007ff801b254e1 libsystem_pthread.dylib`_pthread_start + 125
    frame #12: 0x00007ff801b20f6b libsystem_pthread.dylib`thread_start + 15
```

The crash occurs at this line:

https://github.com/gnuradio/gnuradio/blob/962eb04aaef345bcdc808d98cd4621189237539a/gnuradio-runtime/lib/block_gateway_impl.cc#L59

Unfortunately I don't know much about pybind11, so I'm not sure what's going on here.

willcode, 2022-10-23:
That looks like pybind11 vs macos libsystem problem, a layer below GR. How is pybind11 built for macos? It kind of looks like a dynamic link that didn't get resolved before a function was called.

argilo, 2022-10-23:
On my test system, I installed pybind11 through Homebrew. In CI, it's installed through Conda.

argilo, 2022-10-24:
I see that libgnuradio-runtime is linked to the system Python (3.9.6) rather than the Homebrew Python (3.10.8). Maybe that's the source of the problem.
```
% otool -L /usr/local/lib/libgnuradio-runtime.dylib
/usr/local/lib/libgnuradio-runtime.dylib:
	/usr/local/lib/libgnuradio-runtime.3.11.0git.dylib (compatibility version 3.11.0, current version 0.0.0)
	/usr/local/lib/libgnuradio-pmt.3.11.0git.dylib (compatibility version 3.11.0, current version 0.0.0)
	/usr/local/opt/volk/lib/libvolk.2.5.dylib (compatibility version 2.5.0, current version 2.5.0)
	/usr/local/opt/boost/lib/libboost_program_options-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_thread-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/spdlog/lib/libspdlog.1.dylib (compatibility version 1.0.0, current version 1.10.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.100.3)
	/usr/local/opt/fmt/lib/libfmt.9.dylib (compatibility version 9.0.0, current version 9.1.0)
	/usr/local/opt/gmp/lib/libgmpxx.4.dylib (compatibility version 11.0.0, current version 11.1.0)
	/usr/local/opt/gmp/lib/libgmp.10.dylib (compatibility version 15.0.0, current version 15.1.0)
	/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/Python (compatibility version 3.9.0, current version 3.9.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1300.23.0)
```

```
% ls -alF `which python3`
lrwxr-xr-x  1 argilo  admin  40 20 Oct 18:39 /usr/local/bin/python3@ -> ../Cellar/python@3.10/3.10.8/bin/python3
```

argilo, 2022-10-24:
And all is well after building with `cmake -DPYTHON_EXECUTABLE=/usr/local/bin/python3 ..`. :tada:

I guess all that is needed is to apply the same fix to the Conda build.

argilo, 2022-10-24:
From the Conda build logs, it would appear that cmake is detecting the correct version of Python, so I'm not sure if the same problem is happening there.

argilo, 2022-10-24:
There are, however, two Python versions mentioned in the logs: 3.10.6 and 3.9.13. GNU Radio appears to use 3.9.13.

argilo, 2022-10-24:
@ryanvolz Do you know whether a conflict in Python (or perhaps pybind11) versions could be causing the Python block failures on OSX Conda CI?

ryanvolz, 2022-10-24:
> There are, however, two Python versions mentioned in the logs: 3.10.6 and 3.9.13. GNU Radio appears to use 3.9.13.

Python 3.10.6 is in the base environment, and Python 3.9.13 is in the build/test environments. This is a red herring, everything is fine here and conda is maintaining proper separation.

> @ryanvolz Do you know whether a conflict in Python (or perhaps pybind11) versions could be causing the Python block failures on OSX Conda CI?

Linking to Python is touchy on macOS, and something along this line has come up before with the extension modules:
https://github.com/gnuradio/gnuradio/pull/4057
https://blog.tim-smith.us/2015/09/python-extension-modules-os-x/

Looks like I didn't dig far enough on that.

[`gnuradio-runtime` links with `Python::Python`](https://github.com/gnuradio/gnuradio/blob/main/gnuradio-runtime/lib/CMakeLists.txt#L142), which brings in the Python library symbols which at run time end up competing with the separate Python interpreter used to run the Python block and causing a segfault. This is an issue even if they are identical Python builds.

Can we get away with having `gnuradio-runtime` link to `Python::Module` instead? I assume I tried that back with #4057, but maybe not because I didn't see the need.

ryanvolz, 2022-10-24:
> Can we get away with having `gnuradio-runtime` link to `Python::Module` instead?

And change [`#include <pybind11/embed.h>` to `#include <pybind11/pybind11.h>`](https://github.com/gnuradio/gnuradio/blob/b2c9623cbd548bd86250759007b80b61bd4a2a06/gnuradio-runtime/lib/block_gateway_impl.cc#L11)?

There are a couple more `#include <pybind11/embed.h>` in gr-qtgui, which also links with Python, and maybe that can be removed in the same manner.

ryanvolz, 2022-10-24:
Let's see how these go: https://github.com/ryanvolz/gnuradio/tree/python_no_embed

ryanvolz, 2022-10-24:
Unsurprisingly that didn't work: when not running a Python block, gnuradio-runtime turns up missing Python symbols. I'm at a loss for how to reconcile these competing requirements.

argilo, 2022-10-25:
An observation:

On my laptop (where things now work correctly) both Python an libgnuradio-runtime.dylib are dynamically linked to the same Python library:
```
% otool -L /usr/local/lib/libgnuradio-runtime.dylib
/usr/local/lib/libgnuradio-runtime.dylib:
	/usr/local/lib/libgnuradio-runtime.3.11.0git.dylib (compatibility version 3.11.0, current version 0.0.0)
	/usr/local/lib/libgnuradio-pmt.3.11.0git.dylib (compatibility version 3.11.0, current version 0.0.0)
	/usr/local/opt/volk/lib/libvolk.2.5.dylib (compatibility version 2.5.0, current version 2.5.0)
	/usr/local/opt/boost/lib/libboost_program_options-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_thread-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/spdlog/lib/libspdlog.1.dylib (compatibility version 1.0.0, current version 1.10.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.100.3)
	/usr/local/opt/fmt/lib/libfmt.9.dylib (compatibility version 9.0.0, current version 9.1.0)
	/usr/local/opt/gmp/lib/libgmpxx.4.dylib (compatibility version 11.0.0, current version 11.1.0)
	/usr/local/opt/gmp/lib/libgmp.10.dylib (compatibility version 15.0.0, current version 15.1.0)
	/usr/local/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/Python (compatibility version 3.10.0, current version 3.10.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1300.23.0)

% ls -alF /usr/local/opt/python@3.10
lrwxr-xr-x  1 argilo  admin  28 20 Oct 18:39 /usr/local/opt/python@3.10@ -> ../Cellar/python@3.10/3.10.8

% otool -L `which python3`                         
/usr/local/bin/python3:
	/usr/local/Cellar/python@3.10/3.10.8/Frameworks/Python.framework/Versions/3.10/Python (compatibility version 3.10.0, current version 3.10.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.100.3)
```

On the CI environment, it appears that the Python binary is statically linked:
```
% otool -L $PREFIX/lib/libgnuradio-runtime.dylib
$PREFIX/lib/libgnuradio-runtime.dylib:
	$PREFIX/lib/libgnuradio-runtime.3.11.0git.dylib (compatibility version 3.11.0, current version 0.0.0)
	$PREFIX/lib/libgnuradio-pmt.3.11.0git.dylib (compatibility version 3.11.0, current version 0.0.0)
	@rpath/libvolk.2.5.dylib (compatibility version 2.5.0, current version 2.5.2)
	@rpath/libboost_program_options.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libboost_thread.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libspdlog.1.dylib (compatibility version 1.0.0, current version 1.10.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.0.0)
	@rpath/libfmt.9.dylib (compatibility version 9.0.0, current version 9.1.0)
	@rpath/libgmp.10.dylib (compatibility version 15.0.0, current version 15.1.0)
	@rpath/libpython3.9.dylib (compatibility version 3.9.0, current version 3.9.0)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)

% otool -L $PYTHON
$PREFIX/bin/python:
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1770.106.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.0.0)
```

argilo, 2022-10-25:
Possibly related: https://github.com/pybind/pybind11/issues/3081

ryanvolz, 2022-10-25:
> On the CI environment, it appears that the Python binary is statically linked:

Ah yes, that's why conda is particularly a problem with segfaults.

So perhaps this can be avoided by having `gnuradio-runtime` link to a static `libpython` as well?: https://github.com/ryanvolz/gnuradio/tree/python_static. Not ideal, but it might be the only way out.

argilo, 2022-10-25:
Unfortunately it doesn't look like that worked. The tests involving Python blocks still failed. :cry:

ryanvolz, 2022-10-25:
> Unfortunately it doesn't look like that worked. The tests involving Python blocks still failed. cry

I thought the same, but then I saw that it was still using the dynamic `libpython`. So I still have to figure out the proper CMake incantation (or patch in worst case) to use the static library.

ryanvolz, 2022-10-26:
I got the static `libpython` build to compile, and now that segfaults for all the tests written in Python, not just the ones that use Python blocks. Presumably it's for the same reason of having the Python interpreter conflict with the embedded Python in `gnuradio-runtime`. Back to square one.

ryanvolz, 2022-11-03:
I think not embedding Python into `gnuradio-runtime` is the direction we want to go, as it doesn't seem to me like this should be necessary. [My python_no_embed branch](https://github.com/ryanvolz/gnuradio/tree/python_no_embed) shows that the Python blocks failing on main will pass without the embedding; it's just that `gnuradio-runtime` will fail to load with missing Python symbols for everything that's not a Python block. The pybind11/Python headers are only needed for `gnuradio-runtime` in `block_gateway`, and as far as I can tell `block_gateway` is only used in the Python bindings which then become the basis for Python blocks. If the definition for `block_gateway` could be moved out of `gnuradio-runtime` and into only the `gr_python` Python bindings, I think everything would be happy.

Structurally I don't know how it would be best to do that. Any thoughts along those lines? I could imagine a sort of extension of https://github.com/gnuradio/gnuradio/pull/5444/files, but making `block_gateway` into its own compilation unit, not linking it into `gnuradio-runtime`, and only linking it into `gr_python`. But maybe it would be better to move that code into another location instead of putting it alongside all of `gnuradio-runtime`. Note that doing what I'm describing would change the `gnuradio-runtime` API/ABI, so it would be good to know if there are any direct C++ consumers of `block_gateway` (although I highly doubt it, currently they would be depending on `gnuradio-runtime` built with `ENABLE_PYTHON` anyway).

closed 2022-11-14 by mormj