gnuradio/gnuradio #6313: UHD find_devices_raw missing

jsallay, 2022-10-28:
### What happened?

I would like to run
```
from gnuradio import uhd
uhd.find_devices()
```

The command fails with `AttributeError: module 'gnuradio.uhd.uhd_python' has no attribute 'find_devices_raw'`

Looking at GR3.8, the function was defined in swig.  It looks to me like it wasn't ported over when swig was dropped.  It isn't defined anywhere in the repo.

### System Information

OS: Ubuntu 20.04
GR Installation Method: source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.2

### Steps to Reproduce the Problem

```
from gnuradio import uhd
uhd.find_devices()
```

### Relevant log output

```shell
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/root/.conan/data/gnuradio/3.10.2.0/dev/3-10-2/package/05b62e807d201b9a9b3c32e22513da935e40ec07/lib/python3.8/site-packages/gnuradio/uhd/__init__.py", line 73, in find_devices
    return __builtins__['map'](to_pythonized_dev_addr, uhd_python.find_devices_raw(*args, **kwargs))
AttributeError: module 'gnuradio.uhd.uhd_python' has no attribute 'find_devices_raw'
```

willcode, 2022-10-29:
Does #6159 solve this? That is in v3.10.4.0.

jsallay, 2022-11-01:
That patch solves my immediate problem.  I can search for uhd devices by called `uhd.find("")`  However, the main issue still exists.  In 3.10.4.0, a call to `uhd.find_devices()` still causes the same error.  I'm not sure if the function needs to be removed in favor of `uhd.find` or if the bindings need to be fixed.

willcode, 2022-11-01:
@mbr0wn ^^^

mbr0wn, 2022-11-11:
Yeah I missed reviewing https://github.com/gnuradio/gnuradio/pull/6159 (my fault, not the author or anyone else) but I think we should have used that to fix `find_devices()`, if for API consistency if not anything else. I think there's an easy fix here.