gnuradio/gnuradio #6315: UDP Source & Sink will not work in cross-platform applications

argilo, 2022-10-29:
The UDP Sink directly exposes the in-memory representation of data structures in network packets:

https://github.com/gnuradio/gnuradio/blob/3f9c56abc81f92979760ef8c00dda173a4694f81/gr-network/lib/udp_sink_impl.cc#L197-L215

And the UDP Source does the inverse:

https://github.com/gnuradio/gnuradio/blob/3f9c56abc81f92979760ef8c00dda173a4694f81/gr-network/lib/udp_source_impl.cc#L195-L214

I can see two problems with this:

* If the sender & receiver are running on platforms with different endianness, the `seqnum` and `length` fields will be corrupted.
* I don't think there's any guarantee that `sizeof(header_seq_plus_size)` will be the same on all platforms, so the length of the headers may differ.

I think these blocks should serialize & unserialize headers in a platform-independent manner.

argilo, 2022-10-29:
The payload data is also problematic: the blocks ignore the data type and just `memcpy` the input buffer directly into the packet. For types other than bytes, this wouldn't work between platforms of different endianness.

willcode, 2022-10-29:
I guess I'm sold on making LE the standard for a bunch of things. Adding performance headaches for the 99% (or whatever) systems using LE would be 98% more problematic than taxing the 1%.

RalphSteinhagen, 2022-10-29:
just as additional info: 
[RFC 791](https://www.rfc-editor.org/rfc/rfc791#page-39) seems to be one of the first to reference 'big-endian' as 'network byte order'.

While this is historically a common convention, this is highly debated and some serialiser/deserialisers opt for little-endian (especially for long data arrays) since nowadays most machines are little-endian.

In any case, both are fine as long as the choice is documented if deviating from 'big endian'.