gnuradio/gnuradio #6320: debian/testing RangeWidget causes python exception

march1993, 2022-10-31:
### What happened?

Running uhd_fft and dragging any RangeWidget like RX gain would lead to python exception.

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/qtgui/range.py", line 204, in changed
    self.notifyChanged(self.rangeType(val))
  File "/usr/lib/python3/dist-packages/gnuradio/qtgui/range.py", line 308, in sliderChanged
    self.notifyChanged(self.rangeType(value))
  File "/usr/lib/python3/dist-packages/gnuradio/qtgui/range.py", line 139, in notifyChanged
    self.message_port_pub(self.outputportname_pmt, pmt.cons(self.outputmsgname_pmt, self.value_to_pmt(value)))
AttributeError: 'RangeWidget' object has no attribute 'message_port_pub'
[1]    193592 IOT instruction  uhd_fft -f 90M
FAIL
```

### System Information

Debian testing/bookworm

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

By running any QtGui based executables having RangeWidget

willcode, 2022-10-31:
Replicated this bug on a recent maint-3.10.

argilo, 2022-11-04:
uhd_siggen_gui is broken in the same way.

argilo, 2022-11-04:
It seems to me like we should revert #5566.

Here's my understanding of the situation:

Variables are given special treatment in GRC. Any block which has a `value` property is considered a variable:

https://github.com/gnuradio/gnuradio/blob/3a7e1437bc02b6bf9617a24626a3a0340d52b3a7/grc/core/blocks/block.py#L320-L321

All of the QT GUI blocks that set values (Range, Chooser, Check Box, etc) are variables.
 
Each variable has its `var_make` property rendered into the output:

https://github.com/gnuradio/gnuradio/blob/3a7e1437bc02b6bf9617a24626a3a0340d52b3a7/grc/core/generator/flow_graph.py.mako#L193-L201

Typically these look like so:

https://github.com/gnuradio/gnuradio/blob/3a7e1437bc02b6bf9617a24626a3a0340d52b3a7/gr-qtgui/grc/qtgui_chooser.block.yml#L111

Note that the variable value is assigned to `self.${id}`.

Furthermore, blocks normally have `self.${id} =` prepended to their `make` property when they are rendered, so that the block itself will be stored in `self.${id}`. But variables are excluded from this treatment, because doing so would overwrite the variable value that is already stored there by `var_make`:

https://github.com/gnuradio/gnuradio/blob/3a7e1437bc02b6bf9617a24626a3a0340d52b3a7/grc/core/generator/top_block.py#L209-L214

The `self.${id}` value is used when making connections. Therefore, it is not possible for a variable block to have connections.

A number of new QT GUI blocks were added in #3165, and it seems there was a desire for them to both be variables and have connections. These blocks achieved that by including a `self.${id} = ${win}` statement in their `make` property, overwriting the variable value set by `var_make`. Unfortunately this does not work properly. It allows connections to be made, but up until the first time the widget's value is changed, `self.${id}` will contain a reference to the block, which causes an exception.

The approach taken in #3165 was later applied to the QT GUI Range widget to add a message output in #5566. The resulting exceptions were noticed, and fixed in #6144 by removing `self.${id} = ${win}` from the `make` property, which then broke the ability to make connections. (The message port is broken as of today.)

Given the current design of GRC, I don't see an easy way to make it possible for variable blocks to have connections. So I think it makes sense to revert #5566 and remove the message port. This would also resolve the bugs with uhd_fft and uhf_siggen_gui, which were not updated to take into account the changes that #5566 made to the Range widget's API.

argilo, 2022-11-06:
> uhd_siggen_gui is broken in the same way.

In fact, any Python code generated by GRC prior to #5566 is broken in the same way as well. Moving a range slider generates the `AttributeError: 'RangeWidget' object has no attribute 'message_port_pub'` exception.

closed 2022-11-14 by mormj