gnuradio/gnuradio #6328: D key press deletes blocks rather than disabling them

alexconorrhodes, 2022-11-02:
### What happened?

When I press the D key it deletes blocks, rather than disables. Furthermore, all disabled blocks in my project have been deleted, and when you right click 'disable' it also deletes the blocks!

Thanks,
Alex

### System Information

OS: Windows 11
GR Installation Method: conda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.4.0

### Steps to Reproduce the Problem

Anytime you use the 'disable' function.

marcusmueller, 2022-11-02:
Hi Alex! Thanks for the bug report, well-written!

Good news is that I think that it's unlikely your blocks actually get deleted :) Can you check whether under "View" the "Hide Disabled Blocks" option is turned on?

alexconorrhodes, 2022-11-02:
I'm an idiot - thanks so much!

marcusmueller, 2022-11-02:
You're absolutely **not** an idiot! Had you not written the bug report so precisely, I would have had to guess much more. Such bug reports: Keep 'em coming!

I bet you're aware, but we'd love seeing people on chat.gnuradio.org, or on our discuss-gnuradio@gnu.org mailing list ;)