gnuradio/gnuradio #6337: burst mode with gr-soapy with HackRF sink not working as expected

oe1rsa, 2022-11-04:
### What happened?

I am using message strobe to create a burst PDU, convert it to stream with "PDU to tagged Stream" and sink it to "Soapy HackRF Sink". I initialize burst mode with: self.soapy_hackrf_sink_0.set_length_tag_name("packet_len"). The message strobe is set to about 1 second repeat interval.

I can see that burst mode is activated by a single tPtG appearing in the log messages, but burst mode is not working. It seems like the HackRF buffer is waiting to be filled up until it eventually sends out samples.

I woul have expected instead, that data is sent out as soon as a burst packet is ready, resulting in burst packets sent with a one second interval.

### System Information

OS: Linux Ubuntu 22-04
GR Installation Method: native package and latest main branch from github

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-273-g3f9c56ab

### Steps to Reproduce the Problem

Since it is not obvious to me what would be the "simplest set" for the current problem I will wait until I get some feedback to work out a useful example.

willcode, 2022-11-05:
UHD can be made to work this way, using start/end of burst flags. Not sure that HackRF has support for this.

oe1rsa, 2022-11-05:
Oh, I see. So I will need to dig deeper. I am looking into the SoapyHackRF driver and it looks as if there is something like "burst". At least the word burst appears in several places in the source code. I will see if it can be made to work, however I am not sure where the better place is for changes: gr-soapy or soapy-hackrf ( which in turn depends on libhackrf, oh my).

In either case: Now I can be sure that my intended use case has not been tried before.

willcode, 2022-11-06:
Looking quickly ... tagged bursts should work all the way through from gr-soapy to SoapyHackRF to HackRF. The GRC yml for the deprecated Soapy Sink does have `length_tag_name`, but we did not add that to the simple SoapyHackRF Sink block or the newer Soapy Custom Sink block. It would be trivial to add this tag to those yml files so the param shows up in GRC. First, though, try out the deprecated Soapy Sink block to see if this is the feature you're looking for.

oe1rsa, 2022-11-06:
Ok, I noticed, that the deprecated block has the length_tag_name, however, I managed to set it by "patching" with Python Snippet, by using "Main - After Init" with code:
    self.soapy_hackrf_sink_0.set_length_tag_name("packet_len")
I could see, the setting was effective because the driver responded with a single tPtG message, which is only printed in burst mode.

Still, the burst does not work with hackrf. The likely cause is that the writeStream function of the SoapyHackRF driver does not obey the SOAPY_SDR_END_BURST flag. The only place where the driver makes use of this flag is in the activateStream function.