gnuradio/gnuradio #6345: Bug with DVB Bit Inner Deinterleaver

Electro707, 2022-11-07:
### What happened?

When one tries to run the dvbt_rx_8k example, the example crashes. I have pin-pointed the block to the Bit Inner Deinterleaver.

### System Information

OS: Arch Linux x86_64 
Kernel: 6.0.5-zen1-1-zen
GR Installation: Both from git and the latest released from the package manager

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-298-g3a7e1437

### Steps to Reproduce the Problem

Download the `dvbt_rx_8k` example.
Run it

### Relevant log output

```shell
Thread 18 "dvbt_bit_inner1" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffe97fff6c0 (LWP 203727)]
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44            return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff78a16b3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff7851958 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff783b53d in __GI_abort () at abort.c:79
#4  0x00007ffff62d2112 in std::__glibcxx_assert_fail(char const*, int, char const*, char const*) (file=<optimized out>, line=<optimized out>, function=<optimized out>, condition=<optimized out>) at /usr/src/debug/gcc/libstdc++-v3/src/c++11/debug.cc:60
#5  0x00007ffee529d964 in  () at /usr/lib/libgnuradio-dtv.so.3.11.0git
#6  0x00007fffeb151cef in gr::block_executor::run_one_iteration() () at /usr/lib/libgnuradio-runtime.so.3.11.0git
#7  0x00007fffeb1c981c in gr::tpb_thread_body::tpb_thread_body(std::shared_ptr<gr::block>, std::shared_ptr<boost::barrier>, int) () at /usr/lib/libgnuradio-runtime.so.3.11.0git
#8  0x00007fffeb1b72a0 in  () at /usr/lib/libgnuradio-runtime.so.3.11.0git
#9  0x00007fffeaed9cbb in  () at /usr/lib/libboost_thread.so.1.80.0
#10 0x00007ffff789f8fd in start_thread (arg=<optimized out>) at pthread_create.c:442
#11 0x00007ffff7921a60 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

drmpeg, 2022-11-07:
I'm having a difficult time reproducing this. After many runs, I did see one segfault, but it was in libglib.

Here's the dmesg dump:

```[Mon Nov  7 15:41:04 2022] gnuradio-compan[17622]: segfault at 28 ip 00007f80d7ca5c3c sp 00007f80cfffce70 error 4 in libglib-2.0.so.0.7200.1[7f80d7c6f000+8f000]
[Mon Nov  7 15:41:04 2022] Code: 48 83 ec 08 8b 47 18 85 c0 0f 8e 9f 00 00 00 4c 8b 6b 20 48 89 f5 41 89 d4 4d 85 ed 74 08 4c 89 ef e8 18 05 05 00 48 8b 43 58 <8b> 40 28 85 c0 74 4d 45 85 e4 74 38 48 8b 43 58 48 89 6b 50 44 89```

Electro707, 2022-11-08:
I have been building it and testing it with ArchLinux this whole time, so I will test it tonight in a Debian VM to see if the issue still occurs.

On ArchLinux, I just compiled `maint-3.9` and can confirm this bug isn't on that branch.

I am going to guess the change that breaks it is `7b44a7fe8d216c8e2698493ee40bd11dce016360`. The reason I am guessing instead of trying is because when I tried to build grc on the commit before it, it failed due to the fix in the later PR #5302. I might try spining an Arch VM and have it only use QT5 to allow me to build it with that commit.

drmpeg, 2022-11-08:
Quick update. @argilo may have found the issue. Standby for a fix.

drmpeg, 2022-11-08:
@Electro707 Could you try placing ```if (d_hierarchy != NH)``` in front of line 92 in dvbt_bit_inner_deinterleaver.cc?

https://github.com/gnuradio/gnuradio/blob/main/gr-dtv/lib/dvbt/dvbt_bit_inner_deinterleaver_impl.cc#L92

argilo, 2022-11-08:
Valgrind shows the following buffer overrun:
```
==1552082== Thread 32 dvbt_bit_inner12:
==1552082== Invalid read of size 8
==1552082==    at 0x561A4F88: gr::dtv::dvbt_bit_inner_deinterleaver_impl::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) (dvbt_bit_inner_deinterleaver_impl.cc:92)
==1552082==    by 0xC0E086A: gr::block_executor::run_one_iteration() (block_executor.cc:623)
==1552082==    by 0xC14609A: gr::tpb_thread_body::tpb_thread_body(std::shared_ptr<gr::block>, std::shared_ptr<boost::barrier>, int) (tpb_thread_body.cc:94)
==1552082==    by 0xC12E609: operator() (scheduler_tpb.cc:38)
==1552082==    by 0xC12E609: operator() (thread_body_wrapper.h:49)
==1552082==    by 0xC12E609: __invoke_impl<void, gr::thread::thread_body_wrapper<gr::tpb_container>&> (invoke.h:61)
==1552082==    by 0xC12E609: __invoke_r<void, gr::thread::thread_body_wrapper<gr::tpb_container>&> (invoke.h:111)
==1552082==    by 0xC12E609: std::_Function_handler<void (), gr::thread::thread_body_wrapper<gr::tpb_container> >::_M_invoke(std::_Any_data const&) (std_function.h:290)
==1552082==    by 0xC8700CA: ??? (in /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.74.0)
==1552082==    by 0x4A3FB42: start_thread (pthread_create.c:442)
==1552082==    by 0x4AD0BB3: clone (clone.S:100)
==1552082==  Address 0x5183a138 is 0 bytes after a block of size 8 alloc'd
==1552082==    at 0x4849013: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==1552082==    by 0xC0E2D2A: allocate (new_allocator.h:127)
==1552082==    by 0xC0E2D2A: allocate (alloc_traits.h:464)
==1552082==    by 0xC0E2D2A: _M_allocate (stl_vector.h:346)
==1552082==    by 0xC0E2D2A: std::vector<void*, std::allocator<void*> >::_M_default_append(unsigned long) (vector.tcc:635)
==1552082==    by 0xC0E06B4: resize (stl_vector.h:940)
==1552082==    by 0xC0E06B4: gr::block_executor::run_one_iteration() (block_executor.cc:411)
==1552082==    by 0xC14609A: gr::tpb_thread_body::tpb_thread_body(std::shared_ptr<gr::block>, std::shared_ptr<boost::barrier>, int) (tpb_thread_body.cc:94)
==1552082==    by 0xC12E609: operator() (scheduler_tpb.cc:38)
==1552082==    by 0xC12E609: operator() (thread_body_wrapper.h:49)
==1552082==    by 0xC12E609: __invoke_impl<void, gr::thread::thread_body_wrapper<gr::tpb_container>&> (invoke.h:61)
==1552082==    by 0xC12E609: __invoke_r<void, gr::thread::thread_body_wrapper<gr::tpb_container>&> (invoke.h:111)
==1552082==    by 0xC12E609: std::_Function_handler<void (), gr::thread::thread_body_wrapper<gr::tpb_container> >::_M_invoke(std::_Any_data const&) (std_function.h:290)
==1552082==    by 0xC8700CA: ??? (in /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.74.0)
==1552082==    by 0x4A3FB42: start_thread (pthread_create.c:442)
==1552082==    by 0x4AD0BB3: clone (clone.S:100)
==1552082== 
```
This fixes it:
```diff
diff --git a/gr-dtv/lib/dvbt/dvbt_bit_inner_deinterleaver_impl.cc b/gr-dtv/lib/dvbt/dvbt_bit_inner_deinterleaver_impl.cc
index d750cea821..69c20ed02d 100644
--- a/gr-dtv/lib/dvbt/dvbt_bit_inner_deinterleaver_impl.cc
+++ b/gr-dtv/lib/dvbt/dvbt_bit_inner_deinterleaver_impl.cc
@@ -89,7 +89,9 @@ int dvbt_bit_inner_deinterleaver_impl::general_work(
 {
     const unsigned char* in = (const unsigned char*)input_items[0];
     unsigned char* outh = (unsigned char*)output_items[0];
-    unsigned char* outl = (unsigned char*)output_items[1];
+    unsigned char* outl = NULL;
+    if (d_hierarchy != NH)
+        (unsigned char*)output_items[1];
 
     int bmax = noutput_items * d_nsize / d_bsize;
```

Electro707, 2022-11-08:
Currently building with the fix, will let you know about the result when compiling is complete.
To note that the interleaver also has the same issue, and from what I can tell in the source code it's the same symptom and the same fix.

drmpeg, 2022-11-08:
Noted on the interleaver. There's also some yaml issues that I'll take care of in the fix PR.

Electro707, 2022-11-08:
I just tested it with the deinterleaver with the fix above, and it no longer crashes

closed 2022-11-08 by mormj