gnuradio/gnuradio #6355: Segmentation fault with GNU Radio from Ubuntu PPA and OOT built against Ubuntu repo

daniestevez, 2022-11-11:
### What happened?

gr-satellites has a PPA archive in which the OOT is built against the version of GNU Radio that is in the Ubuntu repository. For Ubuntu 22.04, this is GNU Radio 3.10.1.1. We have noticed that if gr-satellites is installed from this PPA, and GNU Radio 3.10.4.0 from the GNU Radio PPA are both installed on Ubuntu 22.04, then a simple flowgraph that uses one C++ block from gr-satellites segfaults.

More info in https://github.com/daniestevez/gr-satellites/issues/414

I'm afraid of some kind of ABI mismatch, potentially involving pybind11. Is this use case supposed to work, or should we advise people against mixing versions in this way?

### System Information

OS: Ubuntu 22.04
GR Installation Method: GNU Radio 3.10.4.0 installed from Ubuntu PPA. gr-satellites v5.1.0 installed from gr-satellites PPA.

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.4.0

### Steps to Reproduce the Problem

Install GNU Radio from its PPA and gr-satellites from its PPA on Ubuntu 22.04. Use [this flowgraph](https://github.com/ScottTilley/Artemis1/blob/main/phase_wrap_test_copy.grc) (the UHD source can be replaced by a null source). Note that the flowgraph segfaults immediately.

A docker image with this set up, and some gdb info is included in https://github.com/daniestevez/gr-satellites/issues/414

willcode, 2022-11-17:
Versioning is MAJOR.API.ABI.PATCH, so code compiled against 3.10.1.1 is likely to be incompatible with 3.10.4.0. Sometimes the parts of the ABI used by an OOT stay stable for a long time, so this won't always be noticed. There's one ABI for the whole system, not just for `runtime` for example. The gr-satellites PPA should require a specific version (e.g., 3.10.4.X) if possible. A PATCH release is pretty rare these days, too.

daniestevez, 2022-11-17:
That makes sense. I think that it would be easy to modify the gr-satellites `CMakeLists.txt` in the source that is used to build the Ubuntu PPA package to specify a particular ABI version. However, I'm not sure that it would fix the problem. If I recall correctly, the `libgnuradio-satellites.so` is already linked against the appropriate GNU Radio 3.10.1.1 libraries (rather than linking against a generic/unversioned `libgnuradio-runtime.so`, etc.)

I think the problem is that neither Python nor the Ubuntu/Debian packages understand about this versioning. The deb package dependencies make it possible to install gr-satellites from the PPA (which depends on a few libraries from GNU Radio 3.10.1.1) and GNU Radio 3.10.4.0 from the PPA at the same time, simply by adding both PPAs and installing/upgrading both gnuradio and gr-satellites. When this is done, the user now has access to the gr-satellites blocks within gnuradio-companion 3.10.4.0. The generated Python code will do `import satellites`, which happily imports the gr-satellites Python package, because Python knows nothing about ABI versions (I believe). So the Python script is now handling blocks that have been instantiated by GNU Radio 3.10.4.0 code and by GNU Radio 3.10.1.1 code. There is where the crashes happen.

I don't have a good solution for this problem, besides warning users that they should be careful not to mix GNU Radio versions in this way.

Maybe I could try to see how to make the gr-satellites deb package forbid the installation of other versions of GNU Radio via [conflicts](https://www.debian.org/doc/debian-policy/ch-relationships.html#conflicting-binary-packages-conflicts).

willcode, 2022-11-17:
The only solution I can think of that doesn't involve the user building gr-satellites from source is to tie the gr-satellites debs to specific gnuradio versions, and to release gr-satellites-x.y-3.10.4.0 or something along those lines. As long as apt update would do the right thing.

daniestevez, 2023-01-20:
I think this issue can be closed, since warning people not to mix a version of GNU Radio with a .deb package built for a different version of GNU Radio seems the most reasonable solution. Other solutions probably take too much work to implement.

closed 2023-01-20 by daniestevez