gnuradio/gnuradio #6359: double free: gr.prefs.singleton()

marcusmueller, 2022-11-14:
### What happened?

It seems our singleton handling for gr.prefs is fragile; running any python program that instantiates gr.prefs().singleton, or gr.prefs.singleton() results in an abort at program end.

### System Information

OS: Linux (f36) x86_64 glibc 2.35
GR Installation Method: main source build

### GNU Radio Version

3.11-git (main)

### Specific Version

c11667ef6ebbe528289f528bb303d4668732e942

### Steps to Reproduce the Problem

```
python -c 'from gnuradio import gr;gr.prefs.singleton()'
```

for backtrace below, prepend `gdb --args `

### Relevant log output

```shell
#0  0x00007ffff7acbc4c in __pthread_kill_implementation () from /lib64/libc.so.6
#1  0x00007ffff7a7b9c6 in raise () from /lib64/libc.so.6
#2  0x00007ffff7a657f4 in abort () from /lib64/libc.so.6
#3  0x00007ffff7abfd9e in __libc_message () from /lib64/libc.so.6
#4  0x00007ffff7ad595c in malloc_printerr () from /lib64/libc.so.6
#5  0x00007ffff7ad779c in _int_free () from /lib64/libc.so.6
#6  0x00007ffff7ada103 in free () from /lib64/libc.so.6
#7  0x00007fffea012442 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x555555a96b80)
    at /usr/include/c++/12/bits/shared_ptr_base.h:346
#8  std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x555555a96b80) at /usr/include/c++/12/bits/shared_ptr_base.h:317
#9  std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=<optimized out>, __in_chrg=<optimized out>)
    at /usr/include/c++/12/bits/shared_ptr_base.h:1071
#10 std::__shared_ptr<gr::prefs, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=<optimized out>, __in_chrg=<optimized out>)
    at /usr/include/c++/12/bits/shared_ptr_base.h:1524
#11 std::shared_ptr<gr::prefs>::~shared_ptr (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/12/bits/shared_ptr.h:175
#12 pybind11::class_<gr::prefs, std::shared_ptr<gr::prefs> >::dealloc (v_h=...) at /usr/include/pybind11/pybind11.h:1881
#13 0x00007fffe9f74e6d in pybind11::detail::clear_instance (self=0x7fffea0c20b0) at /usr/include/pybind11/detail/class.h:418
#14 0x00007fffe9f7599e in pybind11::detail::pybind11_object_dealloc (self=0x7fffea0c20b0) at /usr/include/pybind11/detail/class.h:442
#15 0x00007ffff7d579bf in _PyEval_EvalFrameDefault () from /lib64/libpython3.10.so.1.0
#16 0x00007ffff7d51293 in _PyEval_Vector () from /lib64/libpython3.10.so.1.0
#17 0x00007ffff7dccd14 in PyEval_EvalCode () from /lib64/libpython3.10.so.1.0
#18 0x00007ffff7dfe0c3 in run_eval_code_obj () from /lib64/libpython3.10.so.1.0
#19 0x00007ffff7df961a in run_mod () from /lib64/libpython3.10.so.1.0
#20 0x00007ffff7df17e1 in PyRun_StringFlags () from /lib64/libpython3.10.so.1.0
#21 0x00007ffff7df16d0 in PyRun_SimpleStringFlags () from /lib64/libpython3.10.so.1.0
#22 0x00007ffff7df07c1 in Py_RunMain () from /lib64/libpython3.10.so.1.0
#23 0x00007ffff7dbcd5b in Py_BytesMain () from /lib64/libpython3.10.so.1.0
#24 0x00007ffff7a66550 in __libc_start_call_main () from /lib64/libc.so.6
#25 0x00007ffff7a66609 in __libc_start_main_impl () from /lib64/libc.so.6
#26 0x0000555555555095 in _start ()
```

marcusmueller, 2022-11-15:
Note: this is a clang  build, maybe this makes a difference. Need to write a unittest to cover this, but writing unit tests that don't run only *within* a python runtime, but that watch how the runtime exits is new to me

willcode, 2022-11-17:
gcc build on Fedora also shows `free(): invalid pointer` `Aborted (core dumped)`

mormj, 2022-12-14:
In the wiki, it advises using `gr.prefs()` to access the preferences object.  Perhaps `gr.prefs.singleton` should not be exposed