gnuradio/gnuradio #6363: osmocom_fft  crashes when frequency range slider/numeric entry widget is adjusted

arrehn, 2022-11-15:
### What happened?

I launched osmocom_fft.py  from the gnuradio package with a hackrf-one connected, and he graphical interface window launched and showed an input signal.

I tried to adjust the center frequency, both by clicking the slider control and by entering a numeric value into the window.
In both cases the application quits unexpectedly.

### System Information

OS: Linux, Windows
GR Installation Method: Tried a bunch, including:

built from github (git, cmake) as of Nov 14, 2022.

Also tested latest radioconda releases for windows, latest pentoo beta liveCD

### GNU Radio Version

3.11-git (main)

### Specific Version

unfortunately I already deleted the folders, do not have exact version info but issue is very easy to repro. Didn't expect I would be the first reporter.

### Steps to Reproduce the Problem

run osmocom_fft.py 
click the frequency control slider.

marcusmueller, 2022-11-15:
Thank you for reporting this!

Osmocom_fft is not part of GNU Radio, but part of the gr-osmosdr Out-Of-Tree module. Since this sounds like it seems most likely connected to the gr-osmosdr hackrf source, I'm afraid this isn't the right place to report it :(

I'd check whether https://osmocom.org/projects/gr-osmosdr/issues already contains a similar issue :)

closed 2022-11-15 by marcusmueller

willcode, 2022-11-17:
This may be fixed by #6342 if it was the same as #6320.