gnuradio/gnuradio #6400: dev-4.0: compiling an OOT fails after commit 39f79f29ad7ec9a605b29d1e901aa694b359453c

dl1ksv, 2022-11-28:
### What happened?

After applying commit 39f79f29ad7ec9a605b29d1e901aa694b359453c compiling an oot fails.

### System Information

Linux F36

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Just compiling the OOT after the commit

### Relevant log output

```shell
[4/10] Compiling C++ object blocklib/funcube/lib/libgnuradio-blocklib-funcube.so.p/.._fcdpp_control_fcdpp_control_cpu.cc.o
FAILED: blocklib/funcube/lib/libgnuradio-blocklib-funcube.so.p/.._fcdpp_control_fcdpp_control_cpu.cc.o 
c++ -Iblocklib/funcube/lib/libgnuradio-blocklib-funcube.so.p -Iblocklib/funcube/lib -I../blocklib/funcube/lib -Iblocklib/funcube/include/gnuradio/funcube -I../blocklib/funcube/include/gnuradio/funcube -Iblocklib/funcube/include -I../blocklib/funcube/include -I../subprojects/nlohmann_json-3.10.5/single_include -Iblocklib/funcube/fcdpp_control -I/usr/local/gnuradio-4.0/include -I/usr/include/python3.10 -I/usr/local/volk/include -I/usr/include -I/usr/include/hidapi -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -std=c++17 -O0 -g -fPIC -DFMT_LOCALE -DFMT_SHARED -DSPDLOG_SHARED_LIB -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DHAVE_CPU -MD -MQ blocklib/funcube/lib/libgnuradio-blocklib-funcube.so.p/.._fcdpp_control_fcdpp_control_cpu.cc.o -MF blocklib/funcube/lib/libgnuradio-blocklib-funcube.so.p/.._fcdpp_control_fcdpp_control_cpu.cc.o.d -o blocklib/funcube/lib/libgnuradio-blocklib-funcube.so.p/.._fcdpp_control_fcdpp_control_cpu.cc.o -c ../blocklib/funcube/fcdpp_control/fcdpp_control_cpu.cc
In Datei, eingebunden von /usr/local/gnuradio-4.0/include/gnuradio/buffer.h:7,
                 von /usr/local/gnuradio-4.0/include/gnuradio/block_work_io.h:3,
                 von /usr/local/gnuradio-4.0/include/gnuradio/block.h:10,
                 von blocklib/funcube/include/gnuradio/funcube/fcdpp_control.h:15,
                 von ../blocklib/funcube/fcdpp_control/fcdpp_control_cpu.h:14,
                 von ../blocklib/funcube/fcdpp_control/fcdpp_control_cpu.cc:11:
/usr/local/gnuradio-4.0/include/gnuradio/tag.h:92:13: Fehler: expected constructor, destructor, or type conversion before »(« token
   92 |     requires(pmtf::is_pmt_derived<PMT_TYPE>::value)
      |             ^
/usr/local/gnuradio-4.0/include/gnuradio/tag.h:131:21: Fehler: »default_tag« bezeichnet keinen Typ
  131 | inline static const default_tag<pmtf::scalar<float>>    SAMPLE_RATE    = { "sample_rate", "Hz", "signal sample rate"};
      |                     ^~~~~~~~~~~
/usr/local/gnuradio-4.0/include/gnuradio/tag.h:132:21: Fehler: »default_tag« bezeichnet keinen Typ
  132 | inline static const default_tag<pmtf::scalar<float>>    SIGNAL_RATE    = SAMPLE_RATE;
      |                     ^~~~~~~~~~~
/usr/local/gnuradio-4.0/include/gnuradio/tag.h:133:21: Fehler: »default_tag« bezeichnet keinen Typ
  133 | inline static const default_tag<pmtf::string>           SIGNAL_NAME    = { "signal_name", "", "signal name"};
      |                     ^~~~~~~~~~~
/usr/local/gnuradio-4.0/include/gnuradio/tag.h:134:21: Fehler: »default_tag« bezeichnet keinen Typ
  134 | inline static const default_tag<pmtf::string>           SIGNAL_UNIT    = { "signal_unit", "", "signal's physical SI unit"};
      |                     ^~~~~~~~~~~
/usr/local/gnuradio-4.0/include/gnuradio/tag.h:135:21: Fehler: »default_tag« bezeichnet keinen Typ
  135 | inline static const default_tag<pmtf::scalar<float>>    SIGNAL_MIN     = { "signal_min", "a.u.", "signal physical max. (e.g. DAQ) limit"};
      |                     ^~~~~~~~~~~
/usr/local/gnuradio-4.0/include/gnuradio/tag.h:136:21: Fehler: »default_tag« bezeichnet keinen Typ
  136 | inline static const default_tag<pmtf::scalar<float>>    SIGNAL_MAX     = { "signal_max", "a.u.", "signal physical max. (e.g. DAQ) limit"};
      |                     ^~~~~~~~~~~
/usr/local/gnuradio-4.0/include/gnuradio/tag.h:137:21: Fehler: »default_tag« bezeichnet keinen Typ
  137 | inline static const default_tag<pmtf::string>           TRIGGER_NAME   = { "trigger_name"};
      |                     ^~~~~~~~~~~
/usr/local/gnuradio-4.0/include/gnuradio/tag.h:138:21: Fehler: »default_tag« bezeichnet keinen Typ
  138 | inline static const default_tag<pmtf::scalar<uint64_t>> TRIGGER_TIME   = { "trigger_time", "ns", "UTC-based time-stamp"};
      |                     ^~~~~~~~~~~
/usr/local/gnuradio-4.0/include/gnuradio/tag.h:139:21: Fehler: »default_tag« bezeichnet keinen Typ
  139 | inline static const default_tag<pmtf::scalar<float>>    TRIGGER_OFFSET = { "trigger_offset", "s", "sample delay w.r.t. the trigger (e.g.compensating analog group delays)"};
      |                     ^~~~~~~~~~~
```

RalphSteinhagen, 2022-11-28:
@dl1ksv strange... :thinking: 

Which compiler/version do you use? I presume F36 == Fedora 36. In the CI/CD this seemed to compile/test [fine](https://github.com/gnuradio/gnuradio/actions/runs/3471963503/jobs/5802153469) for that platform.

The corresponding code is:
https://github.com/gnuradio/gnuradio/blob/716b1a6aa488f7a700b426d382cedeb8005297a8/gr/include/gnuradio/tag.h#L89-L96

which seem to indicate that the `requires` check for a compatible PMT type fails and/or the corresponding template cannot be found (which is in PMTs `type_helpers` which should be (albeit indirectly) available via the `#include <pmtf/wrap.hpp>` on top.

Do you have the means to locally modify the file and re-check your results, e.g. by inserting the `type_helpers.hpp` explicitly and/or making the statement in line 89 `inline static constexpr const char*`...? The inline should help with the linker if it is related to that.

If this doesn't help, maybe @jsallay has further ideas and/or do you have a snapshot to reproduce?

dl1ksv, 2022-11-28:
I use gcc-Version 12.2.1, Fedora 36.
As I build from source , I can modify  the source. I'll try asap.
As mentioned above this commit is the first one my OOT fails to build. Before everything is fine.
If the modification fails, I'll try to upload my oot to git so you can try to reproduce.

RalphSteinhagen, 2022-11-28:
@dl1ksv am not sure whether the mentioned commit is the source of the error (because it works in the CI/CD and locally for Ubuntu/OpenSuse) or whether this triggers an unrelated error. 

I hope that this is just a missing include and or linker 'inline'  ... saw in other OS that they sometimes package/include headers differently. Let me know what you find.

dl1ksv, 2022-11-28:
@RalphSteinhagen Neither adding type_helpers.hpp nor  adding the inline... changed anything.
But it's not a linker but a compiler problem.
I'm curious why compiling of gnuradio does not report a syntax error but only on my oot.

RalphSteinhagen, 2022-11-28:
@dl1ksv I am curious too because in the unit tests these types are explicitly instantiated and work as expected.

Would you have an MVP repo of your OOT that I could have a look at?

jsallay, 2022-11-28:
I can take a look after work today.  Just to be clear.  It looks like this is running on the flatbuffers version - not the variant base version, correct?

RalphSteinhagen, 2022-11-28:
@jsallay thanks. Yes, the variant version isn't merged yet (at least not at that commit) afaik.

jsallay, 2022-11-29:
I haven't run code myself so I can't be sure, but I believe the problem lies in the compilation line.  It has `-std=c++17`.  The concepts used require c++20.  I haven't been following the GR4.0 commits super closely, but I think this might be the first commit that requires c++20.

I am guessing that a simple modification to the meson.build file should fix this for you.

RalphSteinhagen, 2022-11-29:
@dl1ksv thanks for reporting and thanks to @jsallay for spotting this. I updated the modtool accordingly (in existing OOT this would need to be done by hand) and opened a PR.

Unfortunately, the CI/CD doesn't cover OOT use yet. Maybe someone benevolent would find a smart/lean/clean way how to add/create simple OOT within GR 4.0 and write some unit-tests for the modtool...

dl1ksv, 2022-11-29:
First I'd like to thank @RalphSteinhagen and @jsallay for fixing this issue.  Setting cpp_std to c++20 did the trick.
Although gnuradio switched from c++17 -> C++20 the above mentioned commit seems to be  the first one that really requires c++20 features. So it would be great that we could collect such requirements at one point , that could be called inside OOT's , too. In GR3.x  we have a GrCompilerSettings.cmake that can be called , when build OOT's. Changes in the compiler settings for gnuradio can be used in all OOT'S automaticaly.

closed 2022-11-29 by dl1ksv