gnuradio/gnuradio #6415: gnuradio-ofdm-header-modulation

reshma07mr, 2022-12-01:
### What happened?

A bug happened!
I have tried to implement the OFDM transmitter and the receiver, the constellation plot appears same, whichever the header modulation scheme is provided.This is only for the transmitter, for the receiver the selected modulation scheme3 appears in the constellation plot.This condition is same in the Linux and windows Os.

### System Information

OS :Ubuntu 20.04.5 LTS
GR installation method :installed the gnuradio from the git repository.
## gnuradio commands used :
sudo add-apt-repository ppa:gnuradio/gnuradio-releases
sudo apt-get install gnuradio python3-packaging

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

[image]

### Relevant log output

```shell
Nothing
```

willcode, 2022-12-01:
@reshma07mr This report does not include enough information to duplicate your problem. If you have a flowgraph showing a GNU Radio bug, please include it. If you are looking for help with GNU Radio, use the mailing list or chat.gnuradio.org. Feel free to reopen this issue if you have more information.

closed 2022-12-01 by willcode

reshma07mr, 2022-12-05:
Thank you for your response. I was able to manage it. 

But now I have another issue. I had tried to recreate the OFDM tranmsitter sample flowgraph available in the gnuradio/examples/digital/ofdm/tx_ofdm.grc. I tried to plot the constellation plot from different points, but the thing is that, when the output is taken from the repack bits (both from header and payload), it shows that they are not centered at zero. they are shifted towards the positive axis. A sample is attached herealong. Any help or suggestions provided will be appreciated.
[image]

willcode, 2022-12-05:
Please use the mailing list or chat.gnuradio.org for help. This issue tracker is used for bug reports and feature requests.