gnuradio/gnuradio #6427: https://github.com/gnuradio/gnuradio/issues/6415

reshma07mr, 2022-12-05:
Thank you for your response. I was able to manage it. 

But now I have another issue. I had tried to recreate the OFDM tranmsitter sample flowgraph available in the gnuradio/examples/digital/ofdm/tx_ofdm.grc. I tried to plot the constellation plot from different points, but the thing is that, when the output is taken from the repack bits (both from header and payload), it shows that they are not centered at zero. they are shifted towards the positive axis. A sample is attached herealong. Any help or suggestions provided will be appreciated.
[image]

_Originally posted by @reshma07mr in https://github.com/gnuradio/gnuradio/issues/6415#issuecomment-1337437677_

willcode, 2022-12-05:
Please use the mailing list or chat.gnuradio.org for help. The issue tracker is used for bug reports and feature requests.

closed 2022-12-05 by willcode

reshma07mr, 2022-12-05:
Thank you, it is already reported there.