gnuradio/gnuradio #6433: FEC Async Decoder overflows when converting from float to uint8_t

daniestevez, 2022-12-09:
### What happened?

I have noticed that the FEC Async Decoder block can overflow when doing a conversion from `float` to `uint8_t` if the amplitude of the input is too large. This conversion happens, for instance, if the block is used with a CC Decoder Definition to implement a Viterbi decoder.

These are the problematic conversions:
https://github.com/gnuradio/gnuradio/blob/6efdc9d84e6149e3f4049ee7c558dc01106dfd06/gr-fec/lib/async_decoder_impl.cc#L132

https://github.com/gnuradio/gnuradio/blob/6efdc9d84e6149e3f4049ee7c558dc01106dfd06/gr-fec/lib/async_decoder_impl.cc#L185

This is a potential problem when using this block to perform soft-input Viterbi decoding. If the amplitude of the input is too large for any reason, all the conversions to `uint8_t` will overflow, and the effect is that we will pass the inverted symbol stream to the Viterbi decoder. With a transparent convolutional code (most common codes), the Viterbi decoder will happily output the inverted output. This is how I first noticed the problem.

There are some commented lines in the code with calls to `volk_32f_s32f_convert_8i`. These date back to fbbde8d01ab43b7a3ffd5ca9d65cf50a84e2a7be when Tom Rondeau first implemented the block in 2014. The volk kernels for conversion to integers saturate instead of overflowing, but I don't think that the kernels can be used here because there are no kernels that convert to `uint8` (unless we want to jump through some hoops by using the `int8` kernel and adding the bias to convert to `uint8` later, which is not straightforward, because the bias is specified by `d_decoder->get_shift()` as a float, even though that shift would be `128.0` in most cases).

What I think we should do is to use the same code as in the Float to UChar block, since this is exactly the same problem. This block uses [float_array_to_uchar.cc](https://github.com/gnuradio/gnuradio/blob/main/gr-blocks/lib/float_array_to_uchar.cc), which unfortunately I think it is not accessible from gr-fec.

Perhaps we should add a `volk_32f_s32f_s32f_convert_8u` kernel to Volk that includes scale and bias parameters, even if at the moment this kernel would only contain a generic implementation (similar to float_array_to_uchar.cc, but including the multiplication by the scale and addition of the bias).

Thoughts?

### System Information

All systems are affected.

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

This can be tested by feeding an input with too large amplitude (amplitude 4 is already large enough) to a Viterbi decoder implemented using FEC Async Decoder and CC Decoder Defintion, and then looking at the output of the decoder.

japm48, 2022-12-11:
Looks problematic indeed, that conversion leads to UB.
Source: https://timsong-cpp.github.io/cppwp/n3337/conv.fpint#1

marcusmueller, 2023-02-23:
@jdemel um, do you mean GNU Radio's #6537  instead of VOLK's gnuradio/volk#617?

jdemel, 2023-02-23:
Well. I hit the merge button on https://github.com/gnuradio/volk/pull/617 . I had no idea that it would close this issue. This looks like GH automation did something wrong.

willcode, 2023-02-23:
In `#617` the wording `fix gnuradio/gnuradio#6433` caused them to be linked.

daniestevez, 2023-02-23:
Sorry. That was something that GH decided about my wording and I couldn't change (https://github.com/gnuradio/volk/pull/617#issuecomment-1397333512).

closed 2023-03-04 by willcode