gnuradio/gnuradio #6435: Wav file source with libsndfile does not always terminate when done

kng, 2022-12-14:
### What happened?

When using a bunch of .ogg files when processing with gr_satellites using the --wavfile sometimes it is stalls at the end and never exiting properly. This is reproducible with a Wav File Source connected to a Null Sink. 

One [example ogg](https://archive.org/download/satnogs-observations-006850001-006860000/satnogs-observations-006857001-006858000.zip/satnogs_6857031_2022-12-10T09-43-12.ogg) from a SatNOGS observation [6857031](https://network.satnogs.org/observations/6857031/) is available.

Library version `libsndfile1:amd64     1.1.0-3+b1`.
More info in the issue I first posted at [gr_satellites#428](https://github.com/daniestevez/gr-satellites/issues/428)

### System Information

OS: Linux Debian bookworm amd64
GR Installation Method: official debian repos

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

3.10.4.0

### Steps to Reproduce the Problem

Run the flowgraph with the offending ogg, does not exit when finished.

The Wav file source never returns WORK_DONE.

[image]

willcode, 2022-12-16:
I reproduced the problem using the referenced ogg file. Running ogginfo on the file shows

  ...
  `WARNING: EOS not set on stream 1
  ...

Does this happen only on files missing EOS?

luzpaz, 2023-07-13:
> Does this happen only on files missing EOS?

ping @kng

kng, 2023-07-20:
I took another example from those ogg files, same day, same sat: [6854037](https://network.satnogs.org/observations/6854037/) with [ogg](https://archive.org/download/satnogs-observations-006850001-006860000/satnogs-observations-006854001-006855000.zip/satnogs_6854037_2022-12-10T22-34-50.ogg) from the Audio button.

```
$ ogginfo satnogs_6854037_2022-12-10T22-34-50.ogg
Processing file "satnogs_6854037_2022-12-10T22-34-50.ogg"...

New logical stream (#1, serial: 18fb12e2): type vorbis
Vorbis headers parsed for stream 1, information follows...
Version: 0
Vendor: Xiph.Org libVorbis I 20200704 (Reducing Environment)
Channels: 1
Rate: 48000

Nominal bitrate: 239.920000 kb/s
Upper bitrate not set
Lower bitrate not set
User comments section follows...
        ENCODER=satnogs ogg encoder
WARNING: EOS not set on stream 1
Vorbis stream 1:
        Total data length: 8067093 bytes
        Playback length: 7m:07.393s
        Average bitrate: 151.000820 kb/s
```

and the previously trouble-ogg:
```
$ ogginfo satnogs_6857031_2022-12-10T09-43-12.ogg
Processing file "satnogs_6857031_2022-12-10T09-43-12.ogg"...

New logical stream (#1, serial: 12560744): type vorbis
Vorbis headers parsed for stream 1, information follows...
Version: 0
Vendor: Xiph.Org libVorbis I 20180316 (Now 100% fewer shells)
Channels: 1
Rate: 48000

Nominal bitrate: 239.920000 kb/s
Upper bitrate not set
Lower bitrate not set
User comments section follows...
        ENCODER=satnogs ogg encoder
WARNING: EOS not set on stream 1
Vorbis stream 1:
        Total data length: 8774423 bytes
        Playback length: 7m:45.385s
        Average bitrate: 150.832824 kb/s
```

Looks like the same warning on both.
Using those two with the test_ogg results in the 6857031 waiting forever, and 6854037 does complete.
[test_ogg.zip](https://github.com/gnuradio/gnuradio/files/12106378/test_ogg.zip)

What I'm asking perhaps, is that it terminates on EOF or some other method, even if the file is corrupted.
The trouble ogg is one of hundreds that are terminating properly, which is messing up my batch operations quite bad.
Current solution is to convert all ogg to wav which takes up a lot of disk space, but seems to alleviate the problem.

marcusmueller, 2024-02-01:
can, and am more clearly reproducing. Not sure whether this is a GR or libsndfile problem, and that would help know how to fix it

marcusmueller, 2024-02-01:
ah yes, with your file the wavfile_source just produces 0, 0 , 0, … items forever at some point; libsndfile doesn't raise any errors.

I call "this last frame has 0 samples, actually" shenanigans; needs fixing.