gnuradio/gnuradio #6437: Extra Parenthesis in uhd_rfnoc_graph.block.yml

jpalladino, 2022-12-16:
### What happened?

There is a syntax error in an rfnoc flowgraph, due to an error in uhd_rfnoc_graph.block.yml. On line 24 [here](https://github.com/gnuradio/gnuradio/blob/maint-3.10/gr-uhd/grc/uhd_rfnoc_graph.block.yml), there is an extra closing parenthesis:
`      self.rfnoc_graph = ${id} = uhd.rfnoc_graph(uhd.device_addr("${graph_args}")))`
You can see 2 opening parentheses and 3 closing parentheses. 

I noticed this on gr-ettus's github repo and posted an issue [there](https://github.com/EttusResearch/gr-ettus/issues/69), but another user pointed out that the exact same issue is present in uhd_rfnoc_graph.block.yml.

### System Information

OS: Ubuntu 20.04
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.4.0

### Steps to Reproduce the Problem

Create any rfnoc flowgraph using rfnoc_graph and try to run it. A syntax error will be present.

willcode, 2022-12-16:
@mbr0wn this is from a recent PR

mbr0wn, 2023-04-17:
Fixed with 7efa9c6e5a2800bc5ab16fbdeb770b1768f0f08b

closed 2023-04-17 by mbr0wn