gnuradio/gnuradio #6440: Correlate Access Code - Packet Length Mask Error (easy typo fix)

jwkunz, 2022-12-19:
The packet length in this stream is supposed to be 16 bits.  The mask in line 117 of

https://github.com/gnuradio/gnuradio/blob/master/gr-digital/lib/correlate_access_code_bb_ts_impl.cc

currently is 0x0fff.  This mask should be all 16 bits 0xffff.

The completed change should look like this:

int correlate_access_code_bb_ts_impl::header_payload()
{
    return (d_hdr_reg >> 16) & 0xffff;  // Change made here
}

willcode, 2022-12-19:
I can't think of any reason the top 4 bits would be masked off. @marcusmueller do you remember if this is intended?

jwkunz, 2022-12-19:
Could this be a rudimentary protection against requesting a buffer larger than 2^28 bytes?

willcode, 2022-12-20:
There's no buffer, so I don't see any reason for that. Probably it's never been tested with large frames.

jwkunz, 2022-12-20:
Using "large frames" is exactly how I found it in the first place (after several hours of confusion).  Thanks for the quick support.

willcode, 2022-12-20:
There are a couple different header formats defined in GR, and one of them has a 12-bit length. This was probably a cut/paste error, but lets wait until @marcusmueller or other original authors weigh in before changing `correlate_access_code_bb_ts_impl.cc` and `correlate_access_code_ff_ts_impl.cc`.

The blocks that use 12-bit lengths pack the bits, and these blocks do not, so I suspect the 0x0fff is unintended.

mbr0wn, 2024-02-12:
Yeah this is wrong. @marcusmueller and I just checked this together.

PR #7125 (2024-02-13): digital: correlate_access_code TS version: incorrect mask

## Description

Used to be 12 bit mask, but docs and code expect 16 bit.

Also, renamed a private function to be clearer on purpose.

## Related Issue

Fixes #6440

## Which blocks/areas does this affect?

only the TSB version of correlate access code

## Testing Done
closed 2024-02-13 by willcode