gnuradio/gnuradio #6441: Detect when modtool needs pygccxml and doesn't have it

ZeroChaos-, 2022-12-19:
### Feature Description

Per some discussion on #gnuradio (linked to matrix somewhere) I was told that gr_modtool bind doesn't *need* pygccxml all the time.  Unfortunately, if pygccxml is not available and is needed, then the end result is a failure.  It would be awesome if pygccxml not being present could be captured and a message could be shown on `gr_modtool bind` failure which says something to the effect of "This failed, you may wish to try installing pygccxml and try again".  I don't know if this is a reasonable request, but it seems reasonable based on my understanding at the present time.

### Feature Urgency

medium (would be nice to have in the near future)

Aadijindal07, 2023-02-01:
i would like to contribute kindly assign

mustafaadly, 2023-03-27:
Hi !
i tried to work on this by investigating how the pygccxml used and i can see that there is always like an if else statement to check if it exists then use it else use an alternative ,so i wonder if anyone have insights to enlighten me where to start

Aryanth29, 2026-11-08:
Hey, I'm working on this!