gnuradio/gnuradio #6449: QT GUI Frequency Sink in GnuRadio3.10 sometimes shows nothing

doubleelec, 2022-12-27:
### What happened?

When we create such a GRC project like following picture:
[image]
The data file 'test.data' is like that in following zip file:
[test.zip](https://github.com/gnuradio/gnuradio/files/10306940/test.zip)
then, run this project, QT GUI Frequency Sink shows nothing.
However, when we configure file source with 'repeat', the frequency graph would be shown, but I think the data in source file is enough without 'repeat'.
As well, the QT GUI Frequency Sink in GnuRadio 3.9 has no this issue and can show this file anyway.

### System Information

OS: Windows
GR Installation Method: Conda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.5.0

### Steps to Reproduce the Problem

Like I said in 'What happened?', the steps is:
1. Create GRC project like the .png showd;
2. Select the file of file source as 'test.data';
3. Run this project, and the QT GUI Frequency Sink would show nothing.

### Relevant log output

```shell
fft::import_wisdom :error: can't import wisdom from C:\Users\doubl\AppData\Roaming\.gr_fftw_wisdom
pagesize :error: no info; setting pagesize = 4096
```

doubleelec, 2022-12-27:
I guess the relevant log output is not so important, because that log appears on GnuRadio 3.9 also. Howevery I paste it here just for reference only.

willcode, 2023-01-02:
You are seeing the effects of buffering, and probably the flowgraph is terminating before anything is displayed. I think I saw dicussion of this in chat. Closing the issue. Once you get some more experience with GNU Radio, feel free to reopen this issue if you think there is flaw.

closed 2023-01-02 by willcode

doubleelec, 2023-01-04:
> You are seeing the effects of buffering, and probably the flowgraph is terminating before anything is displayed. I think I saw dicussion of this in chat. Closing the issue. Once you get some more experience with GNU Radio, feel free to reopen this issue if you think there is flaw.

Could you explain more why 'the flowgraph is terminating before anything is displayed' and what I can do to avoid that? However, this flowgraph works on GnuRadio 3.9 and failed on GnuRadio 3.10 only.

willcode, 2023-01-04:
Please head over to chat.gnuradio.org or the mailing list for help.