gnuradio/gnuradio #6452: GRC infinite loop and crash: "TypeError: Context.rectangle() takes exactly 4 arguments (0 given)"

paul-k-young, 2023-01-04:
### What happened?

GRC froze when I tried to add the OOT module "QT GUI Inspector Sink" to a new flowgraph.  Additionally the console was flooded with the following messages.

Traceback (most recent call last):
  File "gnuradio\grc\gui\DrawingArea.py", line 223, in draw
    self._flow_graph.draw(cr)
  File "gnuradio\grc\gui\canvas\flowgraph.py", line 535, in draw
    draw_element(cr)
  File "gnuradio\grc\gui\canvas\block.py", line 280, in draw
    port.draw(cr)
  File "gnuradio\grc\gui\canvas\port.py", line 134, in draw
    cr.rectangle(*self._area)
TypeError: Context.rectangle() takes exactly 4 arguments (0 given)

I didn't see any errors related to the Inspector Sink, just the above errors from GRC.

What I expected:

The component to be added to the flowgraph.  Or, if that's not possible, a graceful failure.  Bonus for a sensible error message :)

### System Information

OS: Windows 11
GR Installation Method: Conda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.4.0

### Steps to Reproduce the Problem

Steps to reproduce:
1) Open GNU Radio Companion
2) Select File > New > QT Gui
3) Search for "inspector" in the components list.
4) Drag "GUI > QT GUI Inspector Sink" onto the flowgraph.

### Relevant log output

```shell
Traceback (most recent call last):
  File "C:\Users\paulk\radioconda\lib\site-packages\gnuradio\grc\gui\DrawingArea.py", line 223, in draw
    self._flow_graph.draw(cr)
  File "C:\Users\paulk\radioconda\lib\site-packages\gnuradio\grc\gui\canvas\flowgraph.py", line 535, in draw
    draw_element(cr)
  File "C:\Users\paulk\radioconda\lib\site-packages\gnuradio\grc\gui\canvas\block.py", line 280, in draw
    port.draw(cr)
  File "C:\Users\paulk\radioconda\lib\site-packages\gnuradio\grc\gui\canvas\port.py", line 134, in draw
    cr.rectangle(*self._area)
TypeError: Context.rectangle() takes exactly 4 arguments (0 given)
```

willcode, 2023-01-04:
Some code related to contexts was updated in 3.10.5.0, and that version is available in radioconda. See if upgrading solves  this for you.

paul-k-young, 2023-01-04:
Upgrading didn't fix the issue.

It's probably not relevant but I applied [a patch](https://github.com/gnuradio/gnuradio/issues/2938#issuecomment-1309722026) by @ryanvolz a while back to fix a resize issue. 

I may have pulled in some incompatible packages when upgrading.  I'll try a clean uninstall/install of radioconda to see if that fixes things.

paul-k-young, 2023-01-13:
I did a fresh install of radioconda and still encountered the issue.  I also tried every one of the QT GUI core modules and none of them had an issue.  Since this seems specific to gr-inspector I added  a bug](https://github.com/gnuradio/gr-inspector/issues/47
) in that repo.

I'll leave this bug open since GRC didn't handle the error gracefully and leave it to the maintainers to determine if one or the other should be closed.

paul-k-young, 2023-01-14:
I was able to dig a bit deeper and here's what I noticed for the 'QT GUI Inspector Sink' block.

tl;dr - The block wasn't properly initialized due to one of the canvas param values being an int type whereas the code expected it to be a string type.  This resulted in GRC attempting to repeatedly draw an uninitialized component.

The symptom, an exception in Port.draw(), occurred because the block had not been initialized with Port.create_shapes().  The initialization was failing in Params.truncate() when it was passed an integer param.  (Truncate assumes the input is a string.  See the fix above.)

There was no error printed to the console when the block initialization failed.  Moreover, the program allowed the block to partially load which resulted in GRC repeatedly attempting (and failing) to redraw the component.  Lastly, the behavior could do with some test cases.  (My PR addresses none of these!)

I also haven't figured out whether the Inspector module was mishandling types or if it just happened to be a corner case that wasn't well tested.

marcusmueller, 2024-02-12:
All points to this being a bug in your gr-inspector installation, not GNU Radio! I'm closing this here; but, generally:

- make sure that after you update GNU Radio, you rebuild the OOTs you use
- When facing crashes like these, do try to get a backtrace

closed 2024-02-12 by marcusmueller