gnuradio/gnuradio #6453: dev-4.0: invalid cast when calling getter an enum parameter

frankosterfeld, 2023-01-04:
### What happened?

seen with dev-4.0 (commit 47959775)

I have an enum parameter defined like this:

```
-   id: mode
    label: Sink Mode
    dtype: enums/freq_sink_mode_t
    is_enum: true
    gettable: true
```

The generated getter looks like this:

```
freq_sink_mode_t freq_sink::mode()
{
    return pmtv::cast<freq_sink_mode_t>(request_parameter_query(params::id_mode));
}
```

Calling this method leads to  an exception `Invalid PMT Cast N2gr10digitizers16freq_sink_mode_tE...`

I guess this needs a pmtv::cast<int> which is then cast to freq_sink_mode_t.

### System Information

OS: (Linux Distro, Mac, Windows)
GR Installation Method: (native distro packages - apt, rpm, etc., Conda, Source

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Add gettable enum-type parameter and call the getter.

mormj, 2023-01-05:
Thank you for sharing the issue - looks like you have done a ton of migration over to 4.0 - would love to hear more of your experience and feedback.

I couldn't reproduce locally the issue you were seeing:
https://github.com/mormj/gr4-test_enums

Had to add the `include:` in the yml

NOTE: There were a few other missing includes in pmt library - this needs to be checked in as well for current dev-4.0 OOTs to work more seamlessly (or at all):
https://github.com/gnuradio/pmt/pull/77

frankosterfeld, 2023-01-06:
I've adapted the unit test to reproduce the issue: https://github.com/frankosterfeld/gr4-test_enums/commit/d4095f9e84ed6a3aa3554da12b3cd2b52ed367fd

Basically, when I call mode(), the exception is triggered.

mormj, 2023-01-06:
Ah I see now where that's causing an issue.  Fix incoming.

mormj, 2023-01-09:
Not sure why github isn't automatically closing issues, but PR is now merged

PR #6455 (2023-01-09): 4.0: Couple of pmtv lib merge cleanups
## Description
Couple of things that were missed in the merge of the variant based PMT

## Related Issue
Fixes #6453
related to: https://github.com/gnuradio/gnuradio/pull/6398

## Which blocks/areas does this affect?
non-templated blocks with automatically generated getter functions using enums

## Testing Done
Tested on OOT module with enum parameter

## Checklist

- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [x] I have squashed my commits to have one significant change per commit. 
- [x [... cut]
closed 2023-01-09 by mormj