gnuradio/gnuradio #6456: Constellation Encoder, Decoder, and Soft Decoder ignore symbol mapping

VladislavF, 2023-01-06:
### What happened?

This issue might actually be better stated as only the constellation modulator and receiver respect the symbol mapping defined in the constellation object. 
Essentially for any arbitrary constellation, only the trivial or linear symbol map is guaranteed to work. If you  want to create a grey-coded mapping for instance, the constellation encoder and decoders will not apply the mapping. However the constellation modulator and receiver do in fact respect the mapping. This is very confusing if you've chosen to say manually receive the constellation with discrete RRC filters and sync using the decoder or soft decoder resulting in the incorrect symbols being output even though they were in fact permuted correctly by the transmitter. Worse is if you use both the encoder and decoder, then the symbol map is never applied, and it's not obvious unless you check which symbols are actually sent for which bits.
I've discussed this at some length with @argilo 
In the case of grey-coding, this can be worked around by using the Map block or even permuting the constellation points via something like the following: [points[n ^ n >> 1] for n in range(len(points))]
But this is not proper since mappings can be arbitrary and as they are used by some blocks (Mod&RX), it would make sense for all constellation blocks to respect the definition the same way. This seems to be an issue with the blocks in question (Decode,Soft Decode,Encode) not applying the pre_diff_code() function (please correct me if I'm wrong, recalling from my discussion with Clayton). 

### System Information

OS: WSL
GR Installation Method: pybombs -R gnuradio-default

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.4

### Steps to Reproduce the Problem

Connect a flowgraph as such: constellation encoder->constellation decoder.
Observe: The symbol mapping is not applied by either block (assumes mapping: range(len(constellation)))

See my crude screenshot of the output from a flowgraph as such: vector source->constellation mod.->delay->RRC->constellation decode->GUI time sink

[image]

VladislavF, 2023-01-06:
Actually It looks like the constellation soft decoder may also be handling constellations with higher orders incorrectly. Currently investigating.

VladislavF, 2023-01-16:
The issue in my case was not with the constellation but the automatically generated soft decision LUT using the argument 'auto'.
It seems the only way to get sane soft decisions is to call the python function: digital.soft_dec_table(points, map, precision, Npwr). However, it seems that this function has a few issues. Besides for some reason only being exposed through python, there is an error in the expression here: https://github.com/gnuradio/gnuradio/blob/main/gr-digital/python/digital/soft_dec_lut_gen.py#L199 
distance should be raised to the power of 2. And it's possible noise power must also be scaled by a factor of 2 as mentioned by @daniestevez

ffu, 2025-04-25:
Not sure if this issue is a bug or not, but I agree there is a lot of inconsistency among blocks. Here are some issues to consider:

1. The constellation modulator (digital.generic_mod) has a parameter pre_diff_code to determine whether or not to apply symbol mapping. However, in GRC this is always set to True.

2. The constellation decoder does NOT apply symbol mapping. This is mentioned in the "QPSK Mod and Demod" Tutorial and the recommendation is to use the map block:
 
"This is the hardest part about demodulation. In the synchronization steps, we had basic physics and math on our side. Now, though, we have to interpret some symbol based on what someone else said it was. Basically we just have to know this mapping. And luckily we do, so we use the Map block to convert the symbols from the differential decoder to the original symbols we transmitted."

3. Unlike the constellation decoder, the constellation soft decoder applies symbol mapping. This can certainly trip anyone who transitions from the constellation decoder to the constellation soft decoder.