gnuradio/gnuradio #6463: Optimized bytecode is installed

dvzrv, 2023-01-14:
### What happened?

Hi! I'm currently improving the official gnuradio package for Arch Linux.
On Arch we do not have optimized bytecode files (`.pyo`) (nor support) them in our system packages. AFAIK that's similar on other downstream distributions. 

I currently have to remove them manually after installing them. It would be great if they were not created in the first place :)

### System Information

OS: Arch Linux
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.5.0

### Steps to Reproduce the Problem

```
  local cmake_options=(
    -D CMAKE_INSTALL_PREFIX=/usr
    -D CMAKE_BUILD_TYPE=None
    -D ENABLE_GR_ANALOG=ON
    -D ENABLE_GR_AUDIO=ON
    -D ENABLE_GR_BLOCKS=ON
    -D ENABLE_GR_BLOCKTOOL=ON
    -D ENABLE_GR_CHANNELS=ON
    -D ENABLE_GR_CTRLPORT=ON
    -D ENABLE_GR_DIGITAL=ON
    -D ENABLE_GR_DTV=ON
    -D ENABLE_GR_FEC=ON
    -D ENABLE_GR_FFT=ON
    -D ENABLE_GR_FILTER=ON
    -D ENABLE_GR_IIO=ON
    -D ENABLE_GR_MODTOOL=ON
    -D ENABLE_GR_NETWORK=ON
    -D ENABLE_GR_PDU=ON
    -D ENABLE_GR_QTGUI=ON
    -D ENABLE_GR_SOAPY=ON
    -D ENABLE_GR_TRELLIS=ON
    -D ENABLE_GR_UHD=ON
    -D ENABLE_GR_UTILS=ON
    -D ENABLE_GR_VIDEO_SDL=ON
    -D ENABLE_GR_VOCODER=ON
    -D ENABLE_GR_WAVELET=ON
    -D ENABLE_GR_ZEROMQ=ON
    -D ENABLE_GRC=ON
    -D ENABLE_POSTINSTALL=OFF  # we distribute freedesktop integration directly
    -W no-dev
    -B build
    -S $pkgbase-$pkgver
  )

  cmake "${cmake_options[@]}"
  cmake --build build --verbose
  DESTDIR="$pkgdir" cmake --install build
```

`$pkgdir` will now contain .pyo files.

### Relevant log output

```shell
n/a
```

willcode, 2023-05-05:
@yarda @maitbot Do you install pyo files?

yarda, 2023-05-06:
Yes, we install `.pyo` on Fedora (`python3-gnuradio` rpm package). IMHO the `.pyo` / `.pyc` are prebuilt by the rpm build system https://docs.fedoraproject.org/en-US/packaging-guidelines/Python/#_files_to_include

maitbot, 2023-05-06:
The debhelper Debian Python packaging tools handle not installing .pyo and .pyc files as part of the distributed .deb - as well as a postinst script that does all the byte-compiling on the system when the package is installed.
I would hope that other distributions would have similar Python packaging tools and policies.

In these post-PEP 488 times there are no more .pyo files.
https://peps.python.org/pep-0488/

Also I have a debian/gnuradio.bcep (byte-compilation exception patterns) file to keep from trying to compile examples and, more importantly, the modtool templates.

willcode, 2023-05-06:
Thanks for the input. I think we'll call this a packaging decision.

closed 2023-05-06 by willcode