gnuradio/gnuradio #6466: digital::fll_band_edge_cc::set_samples_per_symbol() is not thread safe

boatbod, 2023-01-16:
### What happened?

Calling digital::fll_band_edge_cc::set_samples_per_symbol() in a multi-threaded environment with the flowgraph running may cause a segmentation violation.

set_samples_per_symbol() implementation calls fll_band_edge_cc::design_filter() which may change the size of the d_filter_lower and d_filter_upper vectors which are used in the fll_band_edge_cc::work() method.  In a threaded environment it is possible for work() to be running at the same time as a filter update and this may allow previously freed memory to be accessed if the filter vector gets smaller.

Suggest adding gr::thread::scoped_lock(d_mutex) to the fll_band_edge_cc block to prevent this happening.

### System Information

OS: Ubuntu 20.04 LTS
GR build from sources on "maint-3.10" branch
Based on code inspection this bug appears to affect multiple releases

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.5.0-1-g480f600c

### Steps to Reproduce the Problem

Bug was discovered after much head-scratching by rebuilding and running gnuradio and op25 with an address sanitizer.
While I don't have a specific test-case available, you should be able to reproduce the problem by running a flowgraph at the same time as making calls to set_samples_per_symbol() that increase then decrease the SPS value.

willcode, 2023-01-16:
Thanks. Agree there should be a mutex there, and also in other setters `set_rolloff()` and `set_filter_size()`.

geekyraghav13, 2023-10-01:
add a lock to protect the d_filter_lower and d_filter_upper vectors from concurrent access. The proposed code change is to add gr:::scoped_lock(d_mutex) within the fll_band_edge_cc block to ensure that only one thread can access the filter vectors at a time. This will prevent the race condition and potential segmentation violation.

willcode, 2023-10-01:
That's right, and feel free to submit a PR if you're interested. The lock should be in work() and probably all of the setters.

geekyraghav13, 2023-10-03:
> That's right, and feel free to submit a PR if you're interested. The lock should be in work() and probably all of the setters.

Actually i dont know about this much and terms like PR  and all i am new in github

marcusmueller, 2023-10-03:
@geekyraghav13  well, you proposed what seems to be a reasonable change, so that's why we're encouraging you to send us your proposed patch in the shape of a pull request (PR). But it's alright if you can't fix this bug right now, then someone else has to do it later.

PR #7127 (2024-06-09): FLL bandedge: fix math & thread safety bugs
- digital: FLL band edge QA with noisy signal
- digital: FLL band edge: fix power calculation and thread safety

## Description

## Related Issue

Closes #6466

Found the normalization bug and the numerical accuracy TODO while doing that
 
## Which blocks/areas does this affect?

FLL bandedge

## Testing Done
closed 2024-06-09 by willcode