gnuradio/gnuradio #6477: gr_modtool bind fails with gcc-12

boatbod, 2023-01-19:
### What happened?

While attempting to run gr_modtool bind <block> on a component of an OOT module the tool is failing with multiple c++ errors.  The same command works fine under gcc-11.

gnorbury@vaio:~/op25/op25/gr-op25$ gr_modtool bind decoder_bf
GNU Radio module name identified: op25
In file included from /tmp/tmp3yi889z4.h:1:
In file included from /home/gnorbury/op25.gr310_gcc12/op25/gr-op25/include/gnuradio/op25/decoder_bf.h:26:
In file included from /usr/local/include/gnuradio/block.h:14:
In file included from /usr/include/c++/12/memory:64:
In file included from /usr/include/c++/12/bits/allocator.h:46:
In file included from /usr/include/x86_64-linux-gnu/c++/12/bits/c++allocator.h:33:
/usr/include/c++/12/bits/new_allocator.h:153:6: error: call to '__builtin_operator_delete' selects
      non-usual deallocation function
            _GLIBCXX_OPERATOR_DELETE(_GLIBCXX_SIZED_DEALLOC(__p, __n),
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[snip]

In **blocktool/core/parseheader_generic.py**, changing the cflags from "-std=c++17 -fPIC" to "-std=c++17 -fPIC -fsized-deallocation" appears to correct the problem.

### System Information

OS: Ubuntu 22.10
GR Installation Method: native packages

Have repeated the test scenario using Ubuntu 22.04 using gcc-11 (works ok), then swapped to gcc-12 with everything else remaining unchanged and the bug makes an appearance.

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

v3.10.5.0-1-g480f600c

### Steps to Reproduce the Problem

Have been working with "https://github.com/boatbod/op25" OOT application
With the above repo cloned into the "op25" directory, you can run the gr_modtool as follows:

cd op25/op25/gr-op25
gr_modtool bind decoder_bf

Switching back and forward between gcc-11 and gcc-12 (using update_alternatives) yields either good or broken behavior.
Manually adding the "-fsized-deallocation" option to gr_modtool's cflags as described above eliminates the problem.

### Relevant log output

```shell
gnorbury@vaio:~/op25/op25/gr-op25$ gr_modtool bind decoder_bf
GNU Radio module name identified: op25
In file included from /tmp/tmptit6ewop.h:1:
In file included from /home/gnorbury/op25.gr310_gcc12/op25/gr-op25/include/gnuradio/op25/decoder_bf.h:26:
In file included from /usr/local/include/gnuradio/block.h:14:
In file included from /usr/include/c++/12/memory:64:
In file included from /usr/include/c++/12/bits/allocator.h:46:
In file included from /usr/include/x86_64-linux-gnu/c++/12/bits/c++allocator.h:33:
/usr/include/c++/12/bits/new_allocator.h:153:6: error: call to '__builtin_operator_delete' selects
      non-usual deallocation function
            _GLIBCXX_OPERATOR_DELETE(_GLIBCXX_SIZED_DEALLOC(__p, __n),
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/12/bits/new_allocator.h:103:35: note: expanded from macro
      '_GLIBCXX_OPERATOR_DELETE'
# define _GLIBCXX_OPERATOR_DELETE __builtin_operator_delete
                                  ^
/usr/include/c++/12/bits/alloc_traits.h:496:13: note: in instantiation of member function
      'std::__new_allocator<char>::deallocate' requested here
      { __a.deallocate(__p, __n); }
            ^
/usr/include/c++/12/bits/basic_string.h:293:24: note: in instantiation of member function
[... 541 lines cut ...]
                                  ^
/usr/include/c++/12/bits/alloc_traits.h:496:13: note: in instantiation of member function
      'std::__new_allocator<long>::deallocate' requested here
      { __a.deallocate(__p, __n); }
            ^
/usr/include/c++/12/bits/stl_vector.h:387:9: note: in instantiation of member function
      'std::allocator_traits<std::allocator<long>>::deallocate' requested here
          _Tr::deallocate(_M_impl, __p, __n);
               ^
/usr/include/c++/12/bits/stl_vector.h:366:2: note: in instantiation of member function
      'std::_Vector_base<long, std::allocator<long>>::_M_deallocate' requested here
        _M_deallocate(_M_impl._M_start,
        ^
/usr/include/c++/12/bits/stl_vector.h:526:7: note: in instantiation of member function
      'std::_Vector_base<long, std::allocator<long>>::~_Vector_base' requested here
      vector() = default;
      ^
/usr/include/c++/12/new:166:6: note: non-usual 'operator delete' declared here
void operator delete(void*, std::size_t, std::align_val_t)
~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
Error occurred while running CASTXML xml file does not exist
gnorbury@vaio:~/op25/op25/gr-op25$
```

boatbod, 2023-01-29:
Just for grins I tried making a brand new OOT with a single general block and tried to bind, and gr_modtool fails with the same error, so it's not something specific to my usage of the tool.

The following patch fixes the problem in gnuradio/blocktool/core/parseheader_generic.py:

--- parseheader_generic.py	2023-01-28 21:16:40.804833423 -0500
+++ parseheader_generic.py.fix	2023-01-28 21:10:32.909996468 -0500
@@ -321,7 +321,7 @@
                 compiler='gcc',
                 undefine_symbols=['__PIE__'],
                 define_symbols=self.define_symbols,
-                cflags='-std=c++17 -fPIC')
+                cflags='-std=c++17 -fPIC -fsized-deallocation')
             decls = parser.parse(
                 [self.target_file], xml_generator_config,
                 compilation_mode=parser.COMPILATION_MODE.ALL_AT_ONCE)

marcusmueller, 2024-05-06:
This is very confusing to me, I must admit. At the point at which that error is raised, we barely included anything *but* `<memory>`. This would sound like Ubuntu 22.10's compiler GCC12 was seriously broken; not quite sure hiding that behind automatically generated constructors is a good idea!

I can't really judge the code paths we'd be taking once we introduce the automatically generated deallocators that `-fsized-deallocators` brings.

marcusmueller, 2024-05-06:
After thorough deliberation:

This seems to be an issue specific to a non-LTS version of Ubuntu in a configuration that's unlikely to occur very often in a compilation step done *within* the `modtool bind` command, which could reasonably worked around by not working on that specific version. 

So, while the fix seems easy, I don't think we'll apply it; we'd have to add checks for Ubuntu's gcc12, (the Ubuntu-specific 6.3 MB of patches to GCC make it hard for me to foresee what changes adding that flag brings on other systems, or whether this is really specific to the Ubuntu gcc). I don't think we can currently shoulder that support envelope, I'm very afraid.

marcusmueller, 2024-05-26:
Reopening. This popped up in default Ubuntu24's config.

MarkMoss, 2024-06-02:
This also impacts Debian Bookworm (gcc 12.2.0 and pygccxml 2.2.1).

[gr_modtool bind on debian bookworm.txt](https://github.com/user-attachments/files/15527216/gr_modtool.bind.on.debian.bookworm.txt)

marcusmueller, 2024-06-24:
@boatbod sorry to warm this up after so long, but could you try whether #7408 works for you?

MarkMoss, 2024-06-24:
I've also been following this issue.

I tested commit ebf2fcfb by running `modtool bind` on a block created following the out of tree module example (https://wiki.gnuradio.org/index.php/OutOfTreeModules) on Ubuntu 24.04 with CastXML 0.6.3.
- g++ 9.5.0: Succeeds.
- g++ 10.5.0: Succeeds.
- g++ 11.4.0: Succeeds.
- g++ 12.3.0: Succeeds.
- g++ 13.2.0: Fails, but with fewer errors than at 3a18360c (5 vs. 20+) .  It looks like the 5 remaining errors are also reported among the 20+ at 3a18360c. 
- g++ 14.0.1: Fails after hitting 20 error limit.  It looks like these are the same 20 errors at ebf2fcfb and 3a18360c.

I've attached logs of the g++ 13 and 14 output run on the out-of-tree example module from .

[g++-13.2.0.log](https://github.com/user-attachments/files/15962094/g%2B%2B-13.2.0.log)
[g++-14.0.1.log](https://github.com/user-attachments/files/15962095/g%2B%2B-14.0.1.log)

So, it seems fixed for g++-12, but may be other issues with g++-13 and later.

closed 2024-06-26 by willcode