gnuradio/gnuradio #6481: GR_IIO_LIBAD9361 is not defined for all gr-iio files

mstfkaratas, 2023-01-20:
### What happened?

I tried to use PlutoSDR sink but encountered with following error.

```
Traceback (most recent call last):
  File "/home/mstf/projects/untitled.py", line 145, in <module>
    main()
  File "/home/mstf/projects/untitled.py", line 123, in main
    tb = top_block_cls()
  File "/home/mstf/projects/untitled.py", line 81, in __init__
    self.iio_pluto_sink_0 = iio.fmcomms2_sink_fc32('' if '' else iio.get_pluto_uri(), [True, True], 32768, False)
AttributeError: module 'gnuradio.iio' has no attribute 'fmcomms2_sink_fc32'

>>> Done (return code 1)
```

Upon inspection i found that symbols for fmcomms2_sink/source are not in related .so file.
Commit 297f7923 changed GR_IIO_LIBAD9361 compile definition from PUBLIC to PRIVATE which made this definition not reach to gr-iio/python/iio/bindings/python_bindings.cc

https://github.com/gnuradio/gnuradio/blob/34f3bd2229eef984b38d55b2b40d3856e0b89370/gr-iio/python/iio/bindings/python_bindings.cc#L24-L32

If i make the definition PUBLIC, GR_IIO_LIBAD9631 definition reaches to bindings.cc and resulting .so file contains related functions.

Following diff seems to solve this problem for me. As I am no expert in CMake i didn't opened a pull request before submitting an issue. 

```diff
diff --git a/gr-iio/lib/CMakeLists.txt b/gr-iio/lib/CMakeLists.txt
index facb00fe0..50c9431e0 100644
--- a/gr-iio/lib/CMakeLists.txt
+++ b/gr-iio/lib/CMakeLists.txt
@@ -31,7 +31,7 @@ if(libad9361_SUFFICIENT)
     target_sources(gnuradio-iio PRIVATE fmcomms2_sink_impl.cc fmcomms2_source_impl.cc
                                         pluto_utils.cc)
     target_link_libraries(gnuradio-iio PRIVATE libad9361::ad9361)
-    target_compile_definitions(gnuradio-iio PRIVATE -DGR_IIO_LIBAD9361)
+    target_compile_definitions(gnuradio-iio PUBLIC -DGR_IIO_LIBAD9361)
 endif(libad9361_SUFFICIENT)
 
 if(ENABLE_COMMON_PCH)

```

### System Information

OS: Arch Linux
GR Installation Method: both from git and pacman repository

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.5.0

### Steps to Reproduce the Problem

python -c "from gnuradio import iio;iio.fmcomms2_sink_fc32('' if '' else iio.get_pluto_uri(), [True, True], 32768, False)"

willcode, 2023-01-20:
Replicated. @ryanvolz can you take a look at the fix above?

ryanvolz, 2023-01-21:
That change makes sense, and I think it's the right way to fix it. Sorry about being overzealous in my conversion to `PRIVATE`!

PR #6483 (2023-01-23): iio: cmake: revert libad9361 linkage back to public

## Description
In #6178, a number of unnecessary public linkages were removed, but this one was needed.

## Related Issue
Fixes #6481

## Which blocks/areas does this affect?
iio

## Checklist
- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [x] I have squashed my commits to have one significant change per commit. 
- [x] I [have signed my commits before making this PR](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md#dco-signed)
- [x] My code follows the code style [... cut]
closed 2023-01-23 by willcode