gnuradio/gnuradio #6501: Check migration to #pragma once

marcusmueller, 2023-02-01:
### Feature Description

As @willcode mentioned in #6497, it seems desirable to switch to `#pragma once`, once that becomes universally available.

- Verify that it's actually a strategic goal to have
- Figure out the blockers on the way there (are there any [relevant compilers without support](https://en.wikipedia.org/wiki/Pragma_once#Portability)?)

The conversion itself would be pretty painless.

### Feature Urgency

medium (would be nice to have in the near future)

marcusmueller, 2023-02-22:
@ThomasHabets are you aware of any reason to stay with the classical include guards nowadays?

ThomasHabets, 2023-02-22:
It's nonstandard, so depends if we aim to support a C++ standard, or a set of specific compilers.

As can be seen in the linked bug manual include guards risk copy-paste, but that's the only benefit I'm aware of. Basically if we know it's not wrong, then there's no point to switching to `#pragma once`. Well, aside from the latter not being possible to reintroduce the copy-paste error.

I guess: if `#pragma once` reduces the set of supported compilers or versions, or indeed makes us need to define a list of supported compilers and versions, then in my opinion it's probably not worth it.

Also I don't have a good sense for if anyone is successfully compiling gnuradio on an unsupported compiler, and this would throw a wrench in that machinery.

But maybe for the compilers we care to support it is the case that if they support C++14 then they support this?

Anyway, this is my thoughts. I have no killer feature either way.

mbr0wn, 2023-04-17:
FYI we switched to `#pragma once` in UHD a while back. I think most of the arguments pro-pragma are overrated (like, compiler performance) but it's just simpler and less of a hassle to get right. All relevant compilers support it. It's not like we support (let alone test) every compiler anyway. We could just make the compatibility with `#pragma once` a requirement for compilers we support.

But do we need to change every single file that's already in GNU Radio? I don't see why. We could allow it going forward.