gnuradio/gnuradio #6504: Error building GnuRadio 3.9 on Raspberian 32-bit 9-22-2022

RobertSwirsky, 2023-02-02:
### What happened?

I've built GnuRadio many times on Ubuntu, now I'm trying to build it on Raspberian and there's an issue.

I need to build it because I need to add some custom blocks for a new SDR  -- the CarribouLite SDR, which is RaspberryPi specific. This build failed on an unmodified clone of 3.9 Maint.

```
Scanning dependencies of target blocks_qa_gr_flowgraph.cc
[ 43%] Building CXX object gr-blocks/lib/CMakeFiles/blocks_qa_gr_flowgraph.cc.dir/qa_gr_flowgraph.cc.o
[ 44%] Linking CXX executable blocks_qa_set_msg_handler.cc
/usr/bin/ld: libgnuradio-blocks.so.3.9.8.0: undefined reference to `__atomic_load_8'
/usr/bin/ld: libgnuradio-blocks.so.3.9.8.0: undefined reference to `__atomic_store_8'
collect2: error: ld returned 1 exit status
make[2]: *** [gr-blocks/lib/CMakeFiles/blocks_qa_set_msg_handler.cc.dir/build.make:118: gr-blocks/lib/blocks_qa_set_msg_handler.cc] Error 1
make[1]: *** [CMakeFiles/Makefile2:2357: gr-blocks/lib/CMakeFiles/blocks_qa_set_msg_handler.cc.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 44%] Linking CXX executable blocks_qa_gr_block.cc
/usr/bin/ld: libgnuradio-blocks.so.3.9.8.0: undefined reference to `__atomic_load_8'
/usr/bin/ld: libgnuradio-blocks.so.3.9.8.0: undefined reference to `__atomic_store_8'
collect2: error: ld returned 1 exit status
make[2]: *** [gr-blocks/lib/CMakeFiles/blocks_qa_gr_block.cc.dir/build.make:118: gr-blocks/lib/blocks_qa_gr_block.cc] Error 1
make[1]: *** [CMakeFiles/Makefile2:2328: gr-blocks/lib/CMakeFiles/blocks_qa_gr_block.cc.dir/all] Error 2
[ 44%] Linking CXX executable blocks_qa_gr_flowgraph.cc
/usr/bin/ld: libgnuradio-blocks.so.3.9.8.0: undefined reference to `__atomic_load_8'
/usr/bin/ld: libgnuradio-blocks.so.3.9.8.0: undefined reference to `__atomic_store_8'
collect2: error: ld returned 1 exit status
make[2]: *** [gr-blocks/lib/CMakeFiles/blocks_qa_gr_flowgraph.cc.dir/build.make:118: gr-blocks/lib/blocks_qa_gr_flowgraph.cc] Error 1
make[1]: *** [CMakeFiles/Makefile2:2386: gr-blocks/lib/CMakeFiles/blocks_qa_gr_flowgraph.cc.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
```

libatomic is installed, and I've tried to make CMakeFiles.txt aware of it manually, but that doesn't seem to propagate to all the block makefiles.

                                         

### System Information

OS: Raspberian
GR Installation Method: (native distro packages - apt, rpm, etc., Conda, Source

### GNU Radio Version

3.9  (maint-3.9)

### Specific Version

Can't build it so this doesn't return anything

### Steps to Reproduce the Problem

Install all the prerequisites for GnuRadio build on Raspberian.
git clone --recurse-submodules 
checkout 3.9 maintanance release
mkdir build; cd build; 

cmake -DNEON_SIMD_ENABLE=OFF -DCMAKE_INSTALL_PREFIX=/usr/local -DOTHER_LIBRARIES=libatomic ../
make

willcode, 2023-02-02:
`apt install libatomic1`. If this does not solve the problem, please use chat.gnuradio.org or the mailing list for help.

RobertSwirsky, 2023-02-02:
libatomic / libatomic1 were installed. 
3.8 builds
I'll try the mailing list.

willcode, 2023-02-02:
Thanks - we use the issue tracker for bugs and feature requests, so you'll get a broader audience on chat/mail.

RobertSwirsky, 2023-02-02:
(Though this is probably a "bug" because the CMake system should have set this up correctly)

willcode, 2023-02-02:
Could be, we can re-open if it debugs down to a bug.

willcode, 2023-02-02:
Related: https://gitlab.kitware.com/cmake/cmake/-/issues/21174

willcode, 2023-02-02:
Let's re-open and make it a feature request. Sounds like it is related to older compilers ... but if someone is interested ...

marcusmueller, 2023-02-08:
yeah, I vaguely remember this on some strangely-configured compilers. "Raspberian" doesn't ring a bell for me; @RobertSwirsky do you per chance mean the original *Raspbian* (i.e., the 32bit debian port that was there before RPi foundation's "Raspberry Pi OS", which was community-maintained)?

In that case: what Board do you actually use, and: is updating to a less um, well-aged operating system an option? (also, the output of `gcc -v` would be interesting.)

RobertSwirsky, 2023-02-08:
Ha! Sorry for my imprecision,  @marcusmueller. I meant, of course:

_Raspberry Pi OS with desktop
Release date: September 22nd 2022
System: 32-bit
Kernel version: 5.15
Debian version: 11 (bullseye)_

and was using the version of gcc supplied with that distribution (after all standard upgrades have been done).

I've build gnuradio many times, and initially thought it was just a missing libatomic1 or I can add a line to the makefile to force it in the link; I wasn't able to.

I've since switched to 

_Raspberry Pi OS with desktop
Release date: September 22nd 2022
System: 64-bit
Kernel version: 5.15
Debian version: 11 (bullseye)_

...and it compiles fine.

I'd like to see this compile issue at least noted in the issue log, even if closed, because with the release of the "CaribouLite" SDR hat for the Raspberry Pi, you're going to see a lot of people trying to build GnuRadio for the Pi and hitting the same wall. At least now they'll know to give up and switch to the 64-bit OS.

marcusmueller, 2023-02-10:
hey, thanks for the detailed system description and debugging! Yeah, sure, nothing like this should ever get lost :)

I'm currently thinking of an elegant way of testing for said problem. Thing is: I can make a `#include <atomic>`-using example that fails to compile without `-latomic` with both `g++` and `clang++` on both my x86_64 and my armhf QEMU, and compiles with `-latomic`. 

Then, again, `message_strobe_impl` uses atomics, and is *not* linked agains atomic; so, I'm clearly not understanding enough what *exactly* goes wrong (best bet so far: for small data types for which the compilers are able to emit actual atomic instructions, the compiler doesn't use library functions; but this is so far unprovable).

marcusmueller, 2023-02-10:
For later reference: our friends over at Pothos had the same problem:

https://github.com/pothosware/SoapyRTLSDR/blob/master/CheckAtomic.cmake

tgiglia, 2023-04-19:
I am having this exact problem. I am using a 64 bit Raspberry PI OS. Has anybody found a solution?
Here are my details:
 cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

 31%] Linking CXX shared library libgnuradio-blocks.so
[ 31%] Built target gnuradio-blocks
[ 31%] Building CXX object gr-blocks/lib/CMakeFiles/blocks_qa_rotator.cc.dir/qa_rotator.cc.o
[ 31%] Linking CXX executable blocks_qa_rotator.cc
[ 31%] Built target blocks_qa_rotator.cc
[ 31%] Building CXX object gr-blocks/lib/CMakeFiles/blocks_qa_gr_block.cc.dir/qa_gr_block.cc.o
[ 31%] Linking CXX executable blocks_qa_gr_block.cc
/usr/bin/ld: libgnuradio-blocks.so.3.9.8.0: undefined reference to `__atomic_load_8'
/usr/bin/ld: libgnuradio-blocks.so.3.9.8.0: undefined reference to `__atomic_store_8'
collect2: error: ld returned 1 exit status
make[2]: *** [gr-blocks/lib/CMakeFiles/blocks_qa_gr_block.cc.dir/build.make:111: gr-blocks/lib/blocks_qa_gr_block.cc] Error 1
make[1]: *** [CMakeFiles/Makefile2:2125: gr-blocks/lib/CMakeFiles/blocks_qa_gr_block.cc.dir/all] Error 2
make: *** [Makefile:146: all] Error 2