gnuradio/gnuradio #6521: praxis.f has compiler errors and thus generating interpolator taps is impossible

SoftwareToolworks, 2023-02-08:
### What happened?

```
gfortran -c -o praxis.o praxis.f
```

### System Information

OS: Linux 5.15.0-58-generic #64-Ubuntu SMP Thu Jan 5 11:43:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
GNU Fortran (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
GR: 3.10.0.0 from Github

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.0.0

### Steps to Reproduce the Problem

```
cd gr-filter/lib/gen_interpolator_taps

gfortran -c -o praxis.o praxis.f
```
alternatively
```
f77 -c -o praxis.o praxis.f
```

### Relevant log output

```shell
praxis.f:23:72:

   23 |  10      X(I) = INITV(I)
      |                                                                        1
Warning: Fortran 2018 deleted feature: DO termination statement which is not END DO or CONTINUE with label 10 at (1)
praxis.f:37:72:

   37 |  30      OUT(I) = X(I)
      |                                                                        1
Warning: Fortran 2018 deleted feature: DO termination statement which is not END DO or CONTINUE with label 30 at (1)
praxis.f:991:24:

  991 |          CALL FLIN(X2,J,F,F2)
      |                        1
Error: Interface mismatch in dummy procedure ‘f’ at (1): Type mismatch in function result (REAL(8)/REAL(4))
praxis.f:1038:21:

 1038 |    40 CALL FLIN(X2,J,F,F2)
      |                     1
Error: Interface mismatch in dummy procedure ‘f’ at (1): Type mismatch in function result (REAL(8)/REAL(4))
```
```

marcusmueller, 2023-02-08:
gfortran -std=legacy -c -o praxis.o praxis.f

works

marcusmueller, 2023-02-08:
@SoftwareToolworks would you like to try my fixes in the linked-above PR?

SoftwareToolworks, 2023-02-08:
> ```
> gfortran -std=legacy -c -o praxis.o praxis.f
> ```
> 
> works

This makes me mad because I tried that. Obviously I had another issue in spelling legacy maybe. Grr.

SoftwareToolworks, 2023-02-08:
I have tried your fixes to
```
generate.sh
```
...and it resolves this issue. Successful compiles and output.

Recommend this issue be closed.

Thanks!

SoftwareToolworks, 2023-02-08:
> @SoftwareToolworks would you like to try my fixes in the linked-above PR?

Your changes worked and solved the problem, and looks a lot cleaner in generate.sh.

SoftwareToolworks, 2023-02-08:
Just one more note, I tried:
```
gfortran -freal-4-real-8 -c -o praxis.o praxis.f
```

and that worked also.

SoftwareToolworks, 2023-02-09:
Solution created

PR #6522 (2023-02-15): filter: make gen_interpolator_taps run on modern gfortran
also, fix C to actually be standards-compliant

also, a .gitignore for compilation results

## Description

We've got some, um, classic FORTRAN in-tree that was/is used to generate interpolator taps.

With these changes, that builds on modern platforms.

## Related Issue

Fixes #6521

## Which blocks/areas does this affect?

Effectively, none. (unless you want to be able to reconstruct how the interpolator taps were calculated!)

## Testing Done
closed 2023-02-09 by SoftwareToolworks