gnuradio/gnuradio #6547: Polyphase Synthesizer crashes on archlinux

alangecker, 2023-02-22:
### What happened?

after porting a script to gnuradio 3.10 the Polyphase Synthesizer, which used to work with these settings, just crashes with `Assertion '__n < this->size()' failed`

```
/usr/include/c++/12.2.1/bits/stl_vector.h:1123: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = float; _Alloc = volk::alloc<float>; reference = float&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    2320650 IOT instruction (core dumped)  python3 
```

weirdly this seems, independently on the version, just crashes on archlinux and is working with every other distribution I've tested (ubuntu & debian)

### System Information

#### crashing setups
3.10.5.0 @ latest archlinux (via pacman)
3.10.4.0 @ latest archlinux (via arch linux archive)
3.10.2.0 @ latest archlinux (via arch linux archive)

#### working setups
3.10.1.1 @ ubuntu 22.04 (via apt)
3.10.3.0 @ ubuntu 22.10 (via apt)
3.10.5.1 @ debian testing (via apt)

### GNU Radio Version

3.10 (maint-3.10)

### Steps to Reproduce the Problem

- have gnuradio 3.10 installed on archlinux
- run this script
```python
from gnuradio import filter
from gnuradio.filter import firdes
from gnuradio.fft import window

num_banks = 30
mod_rate = 200000
taps = firdes.low_pass_2(8, num_banks*mod_rate, 120e3, 50e3, 80, window.WIN_BLACKMAN_HARRIS)

filter.pfb_synthesizer_ccf(num_banks, taps, False) # crashes here on arch

print("seems to work")
```

### Relevant log output

```shell
#0  __GI_abort () at abort.c:53

#1  0x00007fba1c8d30e2 in std::__glibcxx_assert_fail (file=file@entry=0x7fba14c0b248 "/usr/include/c++/12.2.1/bits/stl_vector.h", line=line@entry=1123, 
    function=function@entry=0x7fba14c0b8a0 "std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = float; _Alloc = volk::alloc<float>; reference = float&; size_type = long unsigned int]", 
    condition=condition@entry=0x7fba14c0b04e "__n < this->size()") at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/debug.cc:60

#2  0x00007fba14bbcf69 in std::vector<float, volk::alloc<float> >::operator[] (this=<optimized out>, __n=<optimized out>) at /usr/include/c++/12.2.1/bits/stl_vector.h:1121

#3  std::vector<float, volk::alloc<float> >::operator[] (this=<optimized out>, __n=3) at /usr/include/c++/12.2.1/bits/stl_vector.h:1121

#4  gr::filter::kernel::fir_filter_with_buffer_ccf::set_taps (this=this@entry=0x55cc7fa37610, taps=std::vector of length 0, capacity 0) at /usr/src/debug/gnuradio/gnuradio-3.10.5.0/gr-filter/lib/fir_filter_with_buffer.cc:264

#5  0x00007fba14bbd135 in gr::filter::kernel::fir_filter_with_buffer_ccf::fir_filter_with_buffer_ccf (this=<optimized out>, taps=..., this=<optimized out>, taps=...) at /usr/src/debug/gnuradio/gnuradio-3.10.5.0/gr-filter/lib/fir_filter_with_buffer.cc:241

#6  0x00007fba14c086e8 in std::__new_allocator<gr::filter::kernel::fir_filter_with_buffer_ccf>::construct<gr::filter::kernel::fir_filter_with_buffer_ccf, std::vector<float, std::allocator<float> > const&> (__p=<optimized out>, this=0x55cc7fa5c0a0)
    at /usr/include/c++/12.2.1/bits/new_allocator.h:175

#7  std::allocator_traits<std::allocator<gr::filter::kernel::fir_filter_with_buffer_ccf> >::construct<gr::filter::kernel::fir_filter_with_buffer_ccf, std::vector<float, std::allocator<float> > const&> (__p=<optimized out>, __a=...)
    at /usr/include/c++/12.2.1/bits/alloc_traits.h:516

#8  std::vector<gr::filter::kernel::fir_filter_with_buffer_ccf, std::allocator<gr::filter::kernel::fir_filter_with_buffer_ccf> >::emplace_back<std::vector<float, std::allocator<float> > const&>(std::vector<float, std::allocator<float> > const&) [clone .isra.0] (
    this=this@entry=0x55cc7fa5c0a0) at /usr/include/c++/12.2.1/bits/vector.tcc:117

#9  0x00007fba14bfa827 in gr::filter::pfb_synthesizer_ccf_impl::pfb_synthesizer_ccf_impl (this=<optimized out>, numchans=<optimized out>, taps=..., twox=<optimized out>, this=<optimized out>, numchans=<optimized out>, taps=..., twox=<optimized out>)
    at /usr/src/debug/gnuradio/gnuradio-3.10.5.0/gr-filter/lib/pfb_synthesizer_ccf_impl.cc:55

#10 0x00007fba14bfa9dc in gnuradio::make_block_sptr<gr::filter::pfb_synthesizer_ccf_impl, unsigned int&, std::vector<float, std::allocator<float> > const&, bool&> () at /usr/src/debug/gnuradio/gnuradio-3.10.5.0/gnuradio-runtime/lib/../include/gnuradio/sptr_magic.h:57

#11 gr::filter::pfb_synthesizer_ccf::make (numchans=30, taps=std::vector of length 437, capacity 437 = {...}, twox=<optimized out>) at /usr/src/debug/gnuradio/gnuradio-3.10.5.0/gr-filter/lib/pfb_synthesizer_ccf_impl.cc:27

#12 0x00007fba14cb752c in ?? () from /usr/lib/python3.10/site-packages/gnuradio/filter/filter_python.cpython-310-x86_64-linux-gnu.so

#13 0x00007fba14c6f834 in ?? () from /usr/lib/python3.10/site-packages/gnuradio/filter/filter_python.cpython-310-x86_64-linux-gnu.so
```

marcusmueller, 2023-02-23:
#3  std::vector<float, volk::alloc<float> >::operator[] (this=<optimized out>, __n=3) at /usr/include/c++/12.2.1/bits/stl_vector.h:1121

That's an out-of-bounds access. The fact that we're not seeing that on our platforms might be that `_GLIBCXX_DEBUG` is defined on your platform. Might need to make a build later today where that's the case here, too. But it would seem, at first approach, that `gr::filter::kernel::fir_filter_with_buffer_ccf::set_taps` simply doesn't check/adjust vector size prior to setting taps.

marcusmueller, 2023-02-23:
For easier reference later on:
https://github.com/gnuradio/gnuradio/blob/34f3bd2229eef984b38d55b2b40d3856e0b89370/gr-filter/lib/fir_filter_with_buffer.cc#L244-L268

The relevant line is 

https://github.com/gnuradio/gnuradio/blob/34f3bd2229eef984b38d55b2b40d3856e0b89370/gr-filter/lib/fir_filter_with_buffer.cc#L264

meaning that `d_ntaps + d_naligned - 1` might be `< 3 - 1`.

willcode, 2023-02-23:
I just noticed in the stack trace:
`set_taps (this=this@entry=0x55cc7fa37610, taps=std::vector of length 0, capacity 0)` (empty taps) if that helps.

If `d_naligned = 4`, that would give `resize(3)` and the last iteration of the `for` loop would be
` &d_aligned_taps[3][3]`
which isn't valid, even though a zero-length copy won't actually cause a problem.

For `d_ntaps == 1`, `resize(4)` and `copy( ...one tap..., &d_aligned_taps[3][3])` is valid.

thesamesam, 2023-02-24:
> ```
> #3  std::vector<float, volk::alloc<float> >::operator[] (this=<optimized out>, __n=3) at /usr/include/c++/12.2.1/bits/stl_vector.h:1121
> ```
> 
> That's an out-of-bounds access; the vector was empty/zero-size before. The fact that we're not seeing that on our platforms might be that `_GLIBCXX_DEBUG` is defined on your platform. Might need to make a build later today where that's the case here, too. But it would seem, at first approach, that `gr::filter::kernel::fir_filter_with_buffer_ccf::set_taps` simply doesn't check/adjust vector size prior to setting taps.

Or `_GLIBCXX_ASSERTIONS`.

willcode, 2023-02-24:
There are a number of places in this code that `d_taps == 0` makes for strange conditions. I've been unable to figure out why `set_taps()` would be called with zero length taps, so far, in the context of the given example.

willcode, 2023-02-24:
Ah, found it. In `pfb_synthesizer_ccf_impl::pfb_synthesizer_ccf_impl()`
`const std::vector<float> vtaps(0, d_twox * d_numchans);`
should be
`const std::vector<float> vtaps(d_twox * d_numchans, 0);`
I'll put in a PR.

willcode, 2023-02-24:
Also found/fixed the same error in pfb_arb_resampler.cc.

This would make a good task for someone - check that all vector initializers are using the correct order.

PR #6549 (2023-02-28): filter: fix reversed arguments in vector initializer
## Description
This bug was creating a zero-length vector which was being passed to fir_filter_with_buffer_ccf. A zero-length vector makes no sense for that block, and on systems with certain debug settings an out-of-range copy (of zero length) was detected, causing a crash.

## Related Issue
Fixes #6547 

## Which blocks/areas does this affect?
PFB Synthesizer

## Testing Done
Was unable to cause the reported crash on any systems I have.

Placing breakpoints in `set_taps()` showed that the correct vector was being passed after the change.

 [... cut]
closed 2023-02-28 by willcode