gnuradio/gnuradio #6559: ZMQ Enhancements

jwmelto, 2023-02-28:
### Feature Description

This describes a set of changes to the way ZMQ is implemented in GNU Radio that I think would be beneficial. My experience is primarily with the PUB/SUB stream sinks/sources, so it's possible that my understanding has holes in it.

## Stable Protocol
The over-the-wire format sent between a PUB sink and a SUB source varies, depending upon how the sender is configured. This creates a brittle coupling where the receiver must match the sender's configuration. This is not necessary, if we change how messages are serialized.

I think there may have been a misconception about how ZMQ multi-part messages work that led to the idea that they are expensive and should be avoided if possible. In reality, multi-part messages are managed by the ZMQ layer and are delivered _atomically_ to the application. That is, once the first part of a multi-part message is delivered, **all** parts are available. Furthermore, ZMQ manages the network traffic to maximize efficiency, so we shouldn't be as concerned about it.

Therefore, I propose that **ALL** ZMQ messages follow this multi-part protocol:
1. filter part - I don't know if people have done topic-based publishing in GNU Radio; the requirement of the sink to `bind` to the socket (see #4782) might inhibit multiple publishers doing different filterable things. However, if we are going to support it at all (this could be removed entirely), we should do it consistently. The current processing does put the filter string (if defined!) in a separate message, and simply discards it on read, since ZMQ implements the filtering.
2. tags - today, tags are a configurable feature. I would retain this capability, but dedicate a message part to them. If the block chooses not to send tags, then this part would be empty. The negligible overhead of an empty message part is more than worth the de-coupling of requiring the sender and receiver to agree on tag passing. Today, if tags are enabled in the sink, a special block is pre-pended to the payload. That block would be in this part, separating metadata from data
3. payload - the items, as defined by the stream, as implemented today.

The primary benefit of this stable protocol is that misconfiguration between sinks and sources would be eliminated. A well-defined wire protocol could more easily be used by non-GNU Radio blocks.

I prototyped this approach simply using Python blocks. As expected, the performance was somewhat worse than the current approach. Understanding the performance of a production implementation compared to the existing approach would be a prerequisite to accepting this solution.

## Tags
This deserves a sub-section, because there is some inconsistency in the handling of tags in ZMQ blocks.

The interface to (at least) the PUB/SUB blocks includes a "pass tags" parameter. As noted above, this parameter affects the wire protocol of the data. However, it also subverts the meaning of the `tag_propagation_policy` of the block. That is, if a PUB Sink has `tag_propagation_policy == TPP_DONT` then that block should never send tags. Similarly, a SUB_SOURCE with `pass_tags=True` (required to match the sender's protocol) but `tag_propagation_policy == TPP_DONT` will send any tags received.

While this could be addressed as a simple bug-fix in the currently implementation, using `pass_tags` as a simple protocol specification and honoring the `tag_propagation_policy` of the block, it speaks to the confusion of the interface specification and implementation.

## Bind/Connect
Since #4782 was referenced, I would include it in the refactoring of this feature. 

### Feature Urgency

medium (would be nice to have in the near future)

### More Information

The ZMQ blocks are used simplistically today. As GNU Radio increases in maturity and is more widely used, the value of network interconnects between flowgraphs (and flowgraph parts) will find increasing adoption. Improving the ZMQ implementation to support these uses will contribute to the overall maturity and quality of the ecosystem.

For me, the tag issue is a significant limitation of using ZMQ blocks. I have one application that needs tags from the receiver source; another application that is hostile to them. The value of a PUB sink is that I can fan-out the data to both of these applications, but the tag issue has been a major conundrum.

willcode, 2023-02-28:
I like the general idea the 3-part (from the perspective of GR) messages. I was actualy planning to implement the bind/connect part in the near future. That's a smaller change and should be able to go in quickly. If someone else wants to submit this PR, all the better - just let me know so I don't start on it.

Changing the OTW format will require a backward compatibility mode, though.

willcode, 2023-02-28:
4.0 would be the earliest we could totally get rid of the "V1" OTW protocol. Even then, backward compatibility would be nice. There's already zmq code in the dev-4.0 repo, so let's make sure the two implementations don't stray too far apart if you start putting in PRs for 4.0.

willcode, 2023-03-22:
@jwmelto Take a look at #6596.

willcode, 2023-03-28:
Since GR uses 1- or 2- part messages, we could make a V2 protocol that always has 3 parts (topic, pmt metadata, data). As long as this had to be explicitly selected for sinks, it would be backward compatible. Additionally, the source blocks could detect the type of the metadata. I'm starting to use (topic, json metadata, data) for something else. If the sources saw json, they could either ignore the metadata part, or parse it as they would pmt data. This last part would take some thought so we don't open up a whole class of bugs. The various modes (V1, V2 pmt, V2 json) could be explicitly selected to make things "safer".

A basic version of the blocks than can do all this doesn't seem like a huge thing to write.

jwmelto, 2023-03-28:
There is an open question of whether a single wire protocol for all ZMQ blocks makes sense. Note that "topic" is only meaningful for pub/sub, and need not be a separate message part. Also, tags don't make sense for message blocks; only for streams.

Now, you are suggesting a use case for json vs PMT (I'm curious what that use case is, and what C++ library you are using).

Maybe this should be decoupled further?

willcode, 2023-03-28:
Right now, I'm messing with periodic output from Gqrx to support external tools and long recordings of the spectrum. I'm using nlohmann/json, same as SigMF, to put in metadata about freq/rate/... for each line of fft (or other) output. The topics identify the type of data (raw, linear fft, raw fft). This isn't necessarily something we want for GR, but it's getting me thinking.