gnuradio/gnuradio #6574: Block 'Message Pair to Var' doesn't change the variable from input pair

KVasya, 2023-03-12:
### What happened?

As documentation says, the block ''Message Pair to Var" accepts the pair (Variable, Value) and sets the value 'Value' to the variable 'Variable'. Also the block has a field named 'variable'. In reality the block changes the variable designated in it's field  'variable' instead of changing the variable from message pair.

### System Information

OS: Ubuntu 22.04 LTS
GR Installation Method:  Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.1.1

### Steps to Reproduce the Problem

1. Create a ZMQ SUB message source, connect it to the 'Message Pair to Var' block. 
2. Set the field 'variable' of the 'Message Pair to Var' block with the name of a variable existing in the current gr-project, a separate variable may be created for the purpose (say 'variable_0').  
3. From external Python script send a pair of values through ZMQ socket. The variable name should be one of the existing in the project (say 'variable_1', but not 'variable_0').
4. Control the value of the 'variable_0 with the 'Variable to Message' block and print when it changes with 'Message Debug' block.

willcode, 2023-03-12:
The behavior seems to be as documented, though it's a little unintuitive what this block does. The `car` of the PMT is ignored and the `cdr` is passed to the setter callback.

https://wiki.gnuradio.org/index.php/Message_Pair_to_Var

If there is documentation that says something else, could you please point it out?

Setting an arbitrary variable via a message could also be undesirable, and cause even more runtime errors than this block can already cause.

KVasya, 2023-03-12:
Yes, https://wiki.gnuradio.org/index.php/Message_Pair_to_Var 
says clearly that the block parameter 'variable' indicates the variable to be set with a message.
Still, the first sentence 'This block converts an input message pair to the value of a gnuradio variable' 
implies that both parts of the message (pair) are valuable to the block operation. It's naturally to assume that the
first item of the message means the variable to change.
Perhaps the block doesn't need incoming message to be a pair, and single item message would suffice? 
Or, the block should contain a list of variables (instead of single variable). And the value changes if the car is in the list. 

With current realization I need to create as many blocks as many variables I need to control, with separate zmq sockets for each variable. It would be nice to have  a single block with  a single socket, and a list of variables as mentioned above.

willcode, 2023-03-12:
It should be fairly easy to allow the user to specify a list of variable, use one as a default for backward compatibility, check to see that and incoming message "car" is in the list, and call the appropriate callback.