gnuradio/gnuradio #6575: gr-digital: deprecate Packet_Header blocks

duggabe, 2023-03-14:
### What happened?

In attempting to use these four blocks:
* Packet_Header_Generator
* Packet_Header_Generator_(Default)
* Packet_Header_Parser
* Packet_Header_Parser_(Default)

as part of my https://wiki.gnuradio.org/index.php?title=Packet_Communications tutorial, I found that they have several programming and documentation errors. Even if the errors were corrected, they would only produce the same header format as a Protocol Formatter and Protocol Parser using `digital.header_format_crc`. Those blocks (Protocol Formatter and Protocol Parser) are well tested and also have additional capabilities.

RECOMMENDATION:
The subject Packet_Header blocks should be Deprecated with a suggested reference to the Protocol Formatter and Protocol Parser blocks in the Wiki.

### System Information

OS: Linux Distro
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.5.1

### Steps to Reproduce the Problem

I used `str_pkt_12.grc` as a test (https://pastebin.com/cPLcCFbt) with an input file `padded.txt`  (https://pastebin.com/FHLTXaeL). 
Various parameters were tried in the Formatter Object such as:
* `digital.digital_python.packet_header_default(8)` should have produced a default header containing an access key and packet length, but instead produced 8 bytes of zero
* `digital.digital_python.packet_header_default(4,len_key,num_key,8)` should have produced a CRC header but was not correct

### Relevant log output

```shell
Output from `digital.digital_python.packet_header_default(8)`
***** VERBOSE PDU DEBUG PRINT ******
()
pdu length =          8 bytes
pdu vector contents = 
0000: 00 00 00 00 00 00 00 00 
************************************

Output from `digital.digital_python.packet_header_default(4,len_key,num_key,8)`
***** VERBOSE PDU DEBUG PRINT ******
()
pdu length =          4 bytes
pdu vector contents = 
0000: 00 01 05 00 
************************************
```

Notou, 2023-08-08:
Hi Barry,
I definitely agree that we should choose one set of header generator/parser and their associated format object. 
That looks like needless and problematic code duplication. 

That and we need better documentation.

A good example of that is the fact that what you describe as error is actually normal behavior:
`digital.digital_python.packet_header_default(8)` means that you create a header of 8 bits, output as 8 Char samples because, the default value of bits_per_byte is 1. [See there](https://github.com/gnuradio/gnuradio/blob/35f8a0b310bc8b8d6858324cc87b5337446cc215/gr-digital/include/gnuradio/digital/packet_header_default.h#L45)
And in your test, packets are 256 bytes long, so in binary 1 0000 0000. Given that bits are output LSB first, only the 8 lowest bits can be seen, full of zeros. Increasing `header_len` properly shows the desired number.

As for `digital.digital_python.packet_header_default(4,len_key,num_key,8)`, Yeah, there is something wrong there.
It appears that in this case, instead of having packet length in bits 0-11, packet num in bits 12-23, and crc in bits 24-31 as usual, packet num is in bits 16-27, and crc in bits 32-39. So with a 4 bits interval in-between fields.
Looking at the code, it seems unable to output bits from two successive fields in one element. Hence the padding.

Notou, 2023-08-09:
To try and understand the difference between the two sets, I tried to investigate the `header_format` classes.

I went into that through the `header_format_ofdm` first, since I've been playing with OFDM packets lately..
And I found a few buggy elements, or at least weirdly designed, with negative side effects:

- That class derives from `header_format_crc`, reusing the header format and most functions, only adding a computation of the number of OFDM symbols from packet length. It also has (to reflect the `packet_header_ofdm`API I guess) an option to scramble the header (helping with PAPR). That last option doesn't do anything, for the simple reason that the corresponding lines are [commented](https://github.com/gnuradio/gnuradio/blob/a61091f7cd57d76ae312642fbecef1c4267211e3/gr-digital/lib/header_format_ofdm.cc#L89), seemingly ever since the first commit years ago. Trying to reactivate that led me to find the next point.
- The header format for both CRC and OFDM versions are described as the following: 
  - Payload length in the first 12 bits
  - Packet number in bits 12-23
  - CRC in the last 8 bits
This is said to correspond to the format used by the `packet_header_ofdm`and `packet_header_default`, and it does,... except that those two transmit the field bits LSB first, but not `header_format_crc`:
There, the header is transmitted byte per byte, MSB first. Meaning that instead of having, in order, 12 bits of size, 12 bits of num and 8 bits of CRC, one gets: the lowest 8 bits of size, followed by the lowest 4 bits of num, the 4 highest bits of size, the 8 highest bits of num, and finally the CRC 8. Each of those being themselves MSB first.
This does work, but it is a really strange and unintuitive way to format a header. And I'm pretty sure this is what messes up the scrambling system for OFDM.

That being said, the `header_format`family does appear to me like the way forwards, since it's a bit cleaner, and uses a common `header_buffer`API to simplify reading bits.
In my opinion, that API could really use a way to bitswap fields, instead of byteswap. That would be a more logical processing system.

duggabe, 2023-08-09:
@Notou Your comments about the header format are discussed in Issue #6550 as well.
For my own work, I find the `header_format_default` and the 'Correlate Access Code - Tag Stream' to work best, so I would hate to lose that capability.
Thank you for looking into this.

Notou, 2023-08-28:
I've opened a PR fixing the issues mentioned for `header_format_crc` and `header_format_ofdm`.
These don't affect the other format classes.

With that, the only remaining difference I see between the `packet_header` and `header_format` families, is that the Protocol Formatter only works with full bytes, so the `header_format` family can only operate on headers with a bit size multiple of 8.