gnuradio/gnuradio #6586: gnuradio homebrew formula missing dependency libsndfile

eizemazal, 2023-03-18:
### What happened?

After installing gnuradio via homebrew, trying to build simplistic example in gnuradio-companion, I got error message for waveform generator block:

`Value "analog.GR_COS_WAVE" cannot be evaluated:  name 'analog' is not defined`

After some research I found that activating Python enviroronment for gnuradio and trying to import gnuradio.analog leads to error

`ImportError: dlopen(/usr/local/Cellar/gnuradio/3.10.5.1_1/lib/python3.11/site-packages/gnuradio/analog/analog_python.cpython-311-darwin.so, 2): Library not loaded: /usr/local/opt/libsndfile/lib/libsndfile.1.dylib
  Referenced from: /usr/local/Cellar/gnuradio/3.10.5.1_1/lib/libgnuradio-blocks.3.10.5.1.dylib`

After running  `brew install libsndfile`, the error disappeared.

What should be done is to include libsndfile as a dependency for gnuradio formula in homebrew.

### System Information

OS: Mac OS Big Sur 11.1 / Apple M1
GR Installation Method: Homebrew

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.5.1

### Steps to Reproduce the Problem

`brew install gnuradio`
`gnuradio-companion`

Create signal source block. Double click it and see `Value "analog.GR_COS_WAVE" cannot be evaluated:  name 'analog' is not defined`

marcusmueller, 2023-04-08:
Hi! We're not in control of the homebrew packaging! As I'm sadly not even aware of whom to talk to (otherwise, I would have loved to establish a contact), nor have an apple machine to test on, all I can do point you at the place that probably benefits from a change / bug report:

https://github.com/Homebrew/homebrew-core/blob/413b0b8568d2c608e650a340f7ab506ff3e2f050/Formula/gnuradio.rb

marcusmueller, 2023-04-08:
Ah, seems to be fixed as of https://github.com/Homebrew/homebrew-core/commit/cfbbe4eddfc19f25084370953cb029918cf035af

closed 2023-04-08 by marcusmueller