gnuradio/gnuradio #6587: Burst file source using error

djamic, 2023-03-19:
### What happened?

when i using burst file source not working:
But 3.7 version no problem . 

### System Information

OS: (Linux
GR Installation Method:Source

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

3.10.4.0

### Steps to Reproduce the Problem

burst File Source -->Bursts printer
Bursts file source with generated gr-gsm , but any source not working

### Relevant log output

```shell
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
burst_file_source :info: set_min_output_buffer on block 2 to 1024
11011011101100000101001001110000010010001000000011111000111000101110001011100010101110100101000110011001110011110100111110001001011111010100007610000
thread_body_wrapper :error: ERROR thread[thread-per-block[1]: <block bursts_printer(1)>]: pmt_cdr: wrong_type #t
```

willcode, 2023-03-19:
@djamic There have been some changes to PMTs over the years (gr-gsm was last updated 8 years ago). The fix is probably something fairly simple, but gr-gsm is not part of GNU Radio. I'll leave the issue open here just in case it's a GR bug. It is more likely something that needs to be changed in gr-gsm.

willcode, 2023-03-19:
@djamic Are you using https://github.com/bkerler/gr-gsm?

djamic, 2023-03-19:
@willcode,  yes my build gr-gsm https://github.com/bkerler/gr-gsm (DragonOS ) .But I had to try many other old sources. sometimes I tried to solve this error by replacing the codes from the old sources (with socketpdu) by completing pyinstaller and replacing them with the new system. but this is very rude.

djamic, 2023-03-19:
@willcode I tried to use without gr-gsm, but this error is still there, I think it's with gnu-radio.

willcode, 2023-03-19:
@djamic Since the blocks you mentioned are part of gr-gsm, could you describe the errors you found without using gr-gsm blocks?

marcusmueller, 2023-04-08:
Closing this, because it would seem to be in an OOT, and we've not been able to reproduce it without. Please reopen if you have any input on reproduction!

ptrkrysik, 2025-01-20:
This is actually GNU Radio's bug with de-serialization of PMTs from a file.
Here is simple test code (compile it with 'g++ test.cpp -lgnuradio-pmt -o test'):
```
#include <iostream>
#include <sstream>
#include <pmt/pmt.h>
#include <fstream>

void test_pmt_serialization() {

    pmt::pmt_t pdu_header = pmt::make_dict();

    uint8_t buf[10];
    for (int i = 0; i < 10; i++) {
        buf[i] = i + 1;
    }

    pmt::pmt_t blob = pmt::make_blob(buf, 10);
    pmt::pmt_t msg = pmt::cons(pdu_header, blob);

    std::ofstream output_file("out", std::ofstream::binary);

    pmt::serialize(msg, *output_file.rdbuf());
    output_file.close();

    std::ifstream input_file("out", std::ifstream::binary);

    pmt::pmt_t msg_deserialized = pmt::deserialize(*input_file.rdbuf());

    std::cout << msg << std::endl;
    std::cout << msg_deserialized << std::endl;

    input_file.close();
}

int main() {
    test_pmt_serialization();
    return 0;
}
```

After running the result printed on the screen is following:
```
(() . #[1 2 3 4 5 6 7 8 9 10])
(() . #[7 8 9 10 0 0 0 0 0 0])
```

Expected is to get the same PMTs (before serialization and after deserialization) but they are different.

willcode, 2025-01-20:
Unrelated to the original issue. Please submit a new issue.

closed 2025-01-20 by willcode

ptrkrysik, 2025-01-21:
PMT serialization is used by gr-gsm's burst_file_source and burst_file_sink blocks. This GNU Radio's bug that I described here is a direct cause of these blocks (or one of them) not working. So the statement that it is unrelated is logically false. Anyway I'll reopen the issue with better title if this will get us closer to fixing it.

djamic, 2025-01-21:
> PMT serialization is used by gr-gsm's burst_file_source and burst_file_sink blocks. This GNU Radio's bug that I described here is a direct cause of these blocks (or one of them) not working. So the [...]

Thank you for taking this issue seriously.